OpenTracksApp/OpenTracks #1686: Incorrect display in OSM Dashboard

Geier0815, 2023-09-02:
**Describe the bug**
If I open the track with the cardsymbol, there will be a long blue line shown with an arrow at it's end. On some tracks there is under the arrow is a little square shown at others the line starts at this square. This little square is the map. I have to zoom in heavily and then I can see the track I was ridden.
The last track that will be shown correct is from the 18th of june. After that I had a break until 13th of august and this is the first track that shows this error.
The only thing that has changed, beside updates of opentracks and osmdashboard offline, is that I use a MI5 Band for the heart rate since then.
Changing the format of auto exported data to gpx does not make any difference

**To Reproduce**
1. Open a track
2. Click on the little card symbol

If applicable:
[image]

**Technical information**
 - Device: HMD Global Nokia 4.2      Modell:TA-1157
 - OS: Android One 11
 - OpenTracks version: v4.8.1/v4.8.1 Version Code: 5366

dennisguse, 2023-09-02:
Seems like there are locations that are interpreted as 0,0 (aka west of Africa). 
If you export to KML or GPX and open it in another app, does this work?

@pstorch FYI

pstorch, 2023-09-02:
Is it really a trackpoint west of Africa? The track is only 2.4 km long.

To me it looks like the map tiles are not loading correctly. 
@Geier0815 are you using online or offline maps?
There was a bug related to the overdraw factor and the tile cache size. Both settings can be found in the advanced menu.

Geier0815, 2023-09-02:
This is a track recorded in germany and the tour was really only 2.4 km long. If I zoom in, the tour is shown correcly but with this loooong line at it.
Older tracks, as mentioned before, are shown in the right way, with this version of opentracks and osmdashboard OFFLINE.
All of my skateboard-tracks starting and have started indoor without GPS signal. But all up to the 18.06.23 are displaying correctly and all starting after and including 13.08.23 do not.

pstorch, 2023-09-02:
@Geier0815 can you compare with another app, for example OsmAnd, as @dennisguse suggested?
Or can you send us a track?

Geier0815, 2023-09-02:
A track recorded on the 19.08.23 as gpx is showing in OSMand+ correctly. kmz will not open in osmand.
If one of you give me a private mailaddress I will send you an gpx and an kmz thats not working and one kmz that does.
I am not willing to post private data to an open forum, sorry

pstorch, 2023-09-02:
> I am not willing to post private data to an open forum, sorry

Of course: you can reach me via peter.storch@tuta.io

pstorch, 2023-09-02:
@Geier0815 thanks for the example files. Yes, @dennisguse assumption was correct. The erroneous track contains trackpoints with null latitude and longitude which ware translated by OSMDashboard to coordinates 0,0 (west of africa).
So it's a bug in OSMDashboard. I'll move the issue to the OSMDashboard repo.