OpenTracksApp/OpenTracks #1703: are nightly s working?

nutpantz, 2023-09-22:
The last nightly i have listed is 4.8.3-6
My stable is 4.8.3

Is it building and posting correctly to the nightly server?

Or is it just me?
(I don't see any errors on my end and none of my devices show the nightly as updatable)

If there is no update i would think the nightly should match the latest stable.

pstorch, 2023-09-23:
The nightly is currently fixed to this PR: https://github.com/OpenTracksApp/OpenTracks/pull/1652

@dennisguse should we change back to main branch or should we merge in the changes from main to update it and have a new nightly build?

dennisguse, 2023-09-23:
Rebase done :D

closed 2023-09-23 by dennisguse

pstorch, 2023-09-23:
A new nightly is born 🤗

dennisguse, 2023-09-24:
@pstorch nighlty is still on the now deleted idle branch, right?

pstorch, 2023-09-24:
No, I switched it back to main.

dennisguse, 2023-09-25:
The issue seems to have been me. 
I somehow fucked up the merge of the idle PR (merged, but commits not added to main) and this seems to have prevented updates as (guess) the version id was lower than with the idle branch. 

Now I got the nightly from main :)

dennisguse, 2023-12-27:
PS/ found it :D
This condition occurred as the sensor susbsystem was started twice and then shutdown wasn't done properly.