OpenTracksApp/OpenTracks #1709: History list incorrect whilst tracking activity

IAmTheWebb, 2023-09-27:
**Describe the bug**
The previous activity list looks normal until I start tracking an activity, once tracking has started if I return to the list every item is a repeat of the currently in progress run. Clicking these opens the correct activity.
I don't think this happened on previous versions

**To Reproduce**
1. Start tracking.
2. Navigate back to activity list

If applicable:
[image]

**Technical information**
 - Device: Oneplus Nord
 - OS: LineageOS 20
 - OpenTracks 4.8.4

dennisguse, 2023-09-27:
Thanks for the report. 
I fixed this yesterday. 
Will be in v.4.9.0.

closed 2023-09-27 by dennisguse

dennisguse, 2023-09-27:
Here is the commit: https://github.com/OpenTracksApp/OpenTracks/commit/fda9ef9dd7b1d592a2a4a7ba0d74444c3dc5f0ed