OpenTracksApp/OpenTracks #1711: App crash in track selection mode

khao-soi, 2023-09-28:
**Describe the bug**
App crashes when long pressing a second track in selection mode

**To Reproduce**
1. Long press one track to enable selection mode
2. Long press any other track
3. App crashes

### App information
* ID: de.dennisguse.opentracks
* Version: 5399 v4.8.4

### Device information
* Brand: OnePlus
* Device: OnePlus8T
* Model: KB2003
* Id: TQ3A.230805.001
* Product: OnePlus8T

### Firmware
* SDK: 33
* Release: 13
* Incremental: eng.root.20230910.055511

### Cause of error
```java
Exception in thread "main": java.lang.NullPointerException: Attempt to invoke virtual method 'long de.dennisguse.opentracks.ui.TrackListAdapter$ViewHolder.getId()' on a null object reference
	at de.dennisguse.opentracks.ui.TrackListAdapter.getCheckedIds(TrackListAdapter.java:157)
	at de.dennisguse.opentracks.ui.TrackListAdapter.onPrepareActionMode(TrackListAdapter.java:110)
	at androidx.appcompat.app.AppCompatDelegateImpl$ActionModeCallbackWrapperV9.onPrepareActionMode(AppCompatDelegateImpl.java:3017)
	at androidx.appcompat.view.StandaloneActionMode.invalidate(StandaloneActionMode.java:100)
	at de.dennisguse.opentracks.ui.TrackListAdapter$ViewHolder.onClick(TrackListAdapter.java:266)
	at android.view.View.performClick(View.java:7542)
	at android.view.View.performClickInternal(View.java:7519)
	at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
	at android.view.View$PerformClick.run(View.java:29476)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7924)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

virginiaprivacycoalition, 2023-09-29:
I'm having same issue. For me it crashes even if I only have one track selected. `### App information
* ID: de.dennisguse.opentracks
* Version: 5399 v4.8.4

### Device information
* Brand: POCO
* Device: marble
* Model: 23049PCD8G
* Product: marble_global

### Firmware
* SDK: 33
* Release: 13
* Incremental: V14.0.7.0.TMRMIXM

### Cause of error
```java
Exception in thread "main": java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.ui.TrackListAdapter$ViewHolder.setSelected(boolean)' on a null object reference
	at de.dennisguse.opentracks.ui.TrackListAdapter.setAllSelected(TrackListAdapter.java:135)
	at de.dennisguse.opentracks.TrackListActivity.handleContextItem(TrackListActivity.java:434)
	at de.dennisguse.opentracks.TrackListActivity.-$$Nest$mhandleContextItem(Unknown Source:0)
	at de.dennisguse.opentracks.TrackListActivity$1.onClick(TrackListActivity.java:104)
	at de.dennisguse.opentracks.ui.TrackListAdapter.onActionItemClicked(TrackListAdapter.java:116)
	at androidx.appcompat.app.AppCompatDelegateImpl$ActionModeCallbackWrapperV9.onActionItemClicked(AppCompatDelegateImpl.java:3022)
	at androidx.appcompat.view.StandaloneActionMode.onMenuItemSelected(StandaloneActionMode.java:140)
	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:836)
	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:159)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:987)
	at androidx.appcompat.view.menu.MenuPopup.onItemClick(MenuPopup.java:128)
	at android.widget.AdapterView.performItemClick(AdapterView.java:330)
	at android.widget.AbsListView.performItemClick(AbsListView.java:1222)
	at android.widget.AbsListView$PerformClick.run(AbsListView.java:3236)
	at android.widget.AbsListView.onTouchUp(AbsListView.java:4256)
	at android.widget.AbsListView.onTouchEvent(AbsListView.java:3981)
	at androidx.appcompat.widget.DropDownListView.onTouchEvent(DropDownListView.java:236)
	at androidx.appcompat.widget.MenuPopupWindow$MenuDropDownListView.onTouchEvent(MenuPopupWindow.java:140)
	at android.view.View.dispatchTouchEvent(View.java:15067)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3135)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2802)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3148)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2816)
	at android.view.ViewGroup.dispatchTransformedTouchEvent(ViewGroup.java:3148)
	at android.view.ViewGroup.dispatchTouchEvent(ViewGroup.java:2816)
	at android.widget.PopupWindow$PopupDecorView.dispatchTouchEvent(PopupWindow.java:2578)
	at android.view.View.dispatchPointerEvent(View.java:15342)
	at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:7028)
	at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:6798)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:6217)
	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:6274)
	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:6240)
	at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:6405)
	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:6248)
	at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:6462)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:6221)
	at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:6274)
	at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:6240)
	at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:6248)
	at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:6221)
	at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:9400)
	at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:9351)
	at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:9319)
	at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:9539)
	at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:298)
	at android.os.MessageQueue.nativePollOnce(Native Method)
	at android.os.MessageQueue.next(MessageQueue.java:341)
	at android.os.Looper.loopOnce(Looper.java:169)
	at android.os.Looper.loop(Looper.java:300)
	at android.app.ActivityThread.main(ActivityThread.java:8395)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:559)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:954)
```
`

virginiaprivacycoalition, 2023-09-29:
This also happens to me when I press the any of the other menu buttons on that drop-down with a track selected, including Edit.

dennisguse, 2023-09-29:
@lmarz could you take a look?

BosqueRider, 2023-09-29:
I, too, am having this problem.  Interestingly, after the crash error, hitting the back arrow in the app (this app back arrow didn't work in previous versions), not on the device, actually shows the requested aggregated stats results.  When that screen is closed, the app shuts down and has to be restarted.

Attempting to downgrade from v4.8.4 to 4.8.3 or lower didn't work, either.  Failed to install due to an unknown error.

 
[image]
[image]
[image]
[image]
[image]

lmarz, 2023-09-29:
I can reproduce both kinds of crashes. Sadly, I don't have access to my computer for at least a week, so I can't really send in a patch for now.
The first crash happens when `onLongClick` gets executed when selection mode is already activated. To fix this, we could add an if in onLongClick and call onClick when selection mode is true.
I don't really have an exact idea of why the second crash happens, but we are able to mitigate it, if we check for null for the ViewHolder in `setAllSelected`, `getCheckedIds` and `getCheckedPositions`

dennisguse, 2023-09-30:
I guess, this would be the proper solution: https://developer.android.com/develop/ui/views/layout/recyclerview-custom#java

PR #1723 (2023-10-08): TrackList: Fix crashes in selection mode
Fixes: #1711

The long click crash happened, because we didn't check whether we are already in selection mode.

The context menu item crash happened in `setAllSelected`, because item count != view holder count. That was a design error of me that the SparseBooleanArray uses the view holder positions, which change their contents, as the keys. So now I switched to using the trackIds, which are stored as Integers in the database, so we can easily convert them to long and back. This also simplifies the `getSelectedIds` method. The values from `getSelectedPos [... cut]
commit afbe9419 (2023-10-08): TrackList: Fix crashes in selection mode
Fixes: #1711
closed 2023-10-08 by dennisguse