OpenTracksApp/OpenTracks #1714: Attempt to downgrade from v4.84 to v4.8.3 or lower failed - unknown error

BosqueRider, 2023-09-29:
Downgrade attempt from v4.8.4 to v4.8.3 failed using F-Droid.

Experienced this after the crashes documented in Issue 1711 and included this discovery in  comments there.   Have not tried to downgrade previously, so it is not known whether this is a new problem or an isolated incident.  Attached photos show error message as well as device and app info.

[image]
[image]
[image]

dennisguse, 2023-09-29:
Can you check if downgrading for any app via Fdroid works?
For me, this doesn't work at all.

closed 2023-10-01 by dennisguse

BosqueRider, 2023-10-02:
Okay, I tried downgrading Organic Maps a few minutes ago and get the same "unknown error" message, indicating this is an F-Droid issue, or more likely, my phone. Thanks for the reply. If I figure it out, I'll let you know.

Sent with [Proton Mail](https://proton.me/) secure email.

BosqueRider, 2023-10-02:
Oddly enough, I upgraded Organic Maps without any issues, but trying again to downgrade resulted in the unknown error.

Sent with [Proton Mail](https://proton.me/) secure email.

dennisguse, 2023-10-03:
That's what I expected as Android shouldn't allow this.
More infos here: https://developer.android.com/studio/publish/versioning#versioningsettings

pstorch, 2023-10-03:
I also tried to downgrade a different app, but failed. I had to uninstall first the current version. Be careful to uninstall OpenTracks: do a backup before 😉