OpenTracksApp/OpenTracks #1722: open tracks ne retient pas le parcours ou donne des infos erronées

jungleghost, 2023-10-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1.je click sur le bouton pour demarrer
2. l'application se lance
3. Quant je regarde les infos sur mon trajet,  parfois ça correspond , c'est correct, mais parfois rien n' a été enregistré, je peu faire 5 kilometre , il affiche zéro. Parfois la vitesse moyenne est excessive. Parfois le kilometrage est erronée. 
4. Parfois mon trajet allez est correct mais si je refait un trajet en sens inverse, la partie retour n'est pas comptabilisé
5.
6. See error

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: samsung s8
 - OS: [android 9
 - OpenTracks version: [ v4.9.0]

dennisguse, 2023-10-07:
Duplicate of #1711.

dennisguse, 2023-10-07:
Sounds like a GPS or power saving related problem. 
OpenTaracks doesn't crash, right?

closed 2023-10-10 by dennisguse