OpenTracksApp/OpenTracks #1732: Automatically start recording upon movement detection

CodeShakingSheep, 2023-10-11:
**Is your feature request related to a problem? Please describe.**
Often I forget to hit the record button before I do a workout. This leads to a lot of my workouts not being recorded, so that my statistics don't represent my actual activity. 

**Describe the solution you'd like**
I would love to have a setting, for example called "Automatically start recording when moving" based on sensors and GPS position. When the setting is activated recording of workouts should start automatically upon movement detection. 

**Describe alternatives you've considered**
Alternatively, I think that having a notification upon movement detection asking if I want to start a recording would already be helpful. 

**Additional context**

dennisguse, 2023-10-11:
Which sensor should trigger the start?
GPS is not really suited for this as it is too energy consuming.

Which workout are you thinking about?

CodeShakingSheep, 2023-10-12:
I was thinking about different kinds of activities including cycling, running, walking. Regarding sensors I'm not sure. I thought GPS would be suitable because when the user changes position it means he is moving. But if it's too energy consuming, perhaps the step counter would work with a minimum threshold which could be configured by the user himself in the settings. For example, whenever the device recognizes more than 20 steps then automatically start a recording. Do you think that would be feasible to implement?

dennisguse, 2023-10-12:
Do you want to start a new recording every time movement is detected?
And when would it stop?

CodeShakingSheep, 2023-10-12:
Yes, the idea would be to start a new recording every time movement is detected (based on the user setting of the minimum threshold, could also be more steps than 20). Overall, the idea is to automatize activity recording more so that the user doesn't have to think about it every time before a workout. Recording could stop whenever a minimum threshold period of non-movement is detected, e.g. no steps taken / GPS position not changed for 5 minutes. What do you think?

dennisguse, 2023-10-13:
You can try out what happens if you let OpenTracks run continuously for a day and set the location interval to something like 5min.

By the way, what do you consider a workout?
Just walking around is likely not the case, right?

khao-soi, 2023-10-31:
How about OT somehow detects workout as a kind of draft / suggestion which could be converted to an actual track later. 

If movement is detected the recording starts until movement stops for a while. All these movements appear in an additional list to be selected as an actual track. 

This way the bike ride to the grocery store could be ignored, while the actual run in the park can be marked as a track recording. 

Just an idea.

dennisguse, 2024-12-29:
I would recommend to use the Public API and do the detection outside of OpenTracks.

closed 2024-12-29 by dennisguse