OpenTracksApp/OpenTracks #1734: Crash when Sharing track from track list

evnpwr, 2023-10-14:
When I select a track from the track list by long-pressing a track, and then press the share button I get the following crash report. If I press 'back' from the crash report I am still able to share the track with other apps.

### App information
* ID: de.dennisguse.opentracks
* Version: 5428 v4.9.0

### Device information
* Brand: OnePlus
* Device: OnePlusN200
* Model: DE2117
* Id: TQ3A.230901.001
* Product: OnePlusN200

### Firmware
* SDK: 33
* Release: 13
* Incremental: 91168ac2e8

### Cause of error
```java
Exception in thread "main": java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.ui.TrackListAdapter$ViewHolder.setSelected(boolean)' on a null object reference
	at de.dennisguse.opentracks.ui.TrackListAdapter.setAllSelected(TrackListAdapter.java:135)
	at de.dennisguse.opentracks.ui.TrackListAdapter.onDestroyActionMode(TrackListAdapter.java:126)
	at androidx.appcompat.app.AppCompatDelegateImpl$ActionModeCallbackWrapperV9.onDestroyActionMode(AppCompatDelegateImpl.java:3027)
	at androidx.appcompat.view.StandaloneActionMode.finish(StandaloneActionMode.java:110)
	at de.dennisguse.opentracks.ui.TrackListAdapter.onActionItemClicked(TrackListAdapter.java:117)
	at androidx.appcompat.app.AppCompatDelegateImpl$ActionModeCallbackWrapperV9.onActionItemClicked(AppCompatDelegateImpl.java:3022)
	at androidx.appcompat.view.StandaloneActionMode.onMenuItemSelected(StandaloneActionMode.java:140)
	at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:836)
	at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:159)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:987)
	at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:977)
	at androidx.appcompat.widget.ActionMenuView.invokeItem(ActionMenuView.java:625)
	at androidx.appcompat.view.menu.ActionMenuItemView.onClick(ActionMenuItemView.java:156)
	at android.view.View.performClick(View.java:7542)
	at android.view.View.performClickInternal(View.java:7519)
	at android.view.View.-$$Nest$mperformClickInternal(Unknown Source:0)
	at android.view.View$PerformClick.run(View.java:29476)
	at android.os.Handler.handleCallback(Handler.java:942)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:201)
	at android.os.Looper.loop(Looper.java:288)
	at android.app.ActivityThread.main(ActivityThread.java:7924)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:548)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:936)
```

dennisguse, 2023-10-14:
Duplicate of #1711. 
Fix is in v4.9.1.

closed 2023-10-14 by dennisguse

evnpwr, 2023-10-14:
Thanks! Apologies for the duplicate.

dennisguse, 2023-10-14:
No worries :D