OpenTracksApp/OpenTracks #1736: confusing French translation for two different activities

Myoth, 2023-10-15:
both "activity_type_biking" and "activity_type_cycling" have the same translation in french ("cyclisme"), resulting in confusion when selecting an activity. from my understanding of the difference between biking and cycling, the former is more casual than the latter. as such, I propose activty_type_biking to be translated to "vélo" and activity_type_cycling to be kept as "cyclisme".

dennisguse, 2023-10-15:
You may create a PR or use the translation tool.

PS some of these things are ancient history;)

Myoth, 2023-10-23:
I submitted a translation (the translating tool is pretty confusing, so I think I did), I noticed the redundancy was apparently even picked up by the translating tool, my edit seemed to make it happy by turning it from red to green, hopefully this fixes it !

dennisguse, 2023-10-25:
I fetched the new translations and they will be the nightly in 5min as well as in the next release!
Thanks :D

closed 2023-10-25 by dennisguse