OpenTracksApp/OpenTracks #1737: Crash when not using

sgofferj, 2023-10-17:
### App information
* ID: de.dennisguse.opentracks.nightly
* Version: 5483 v4.9.2-17-g8f9ba046c

### Device information
* Brand: ulefone
* Device: Armor_3
* Model: Armor_3
* Id: O11019
* Product: Armor_3

### Firmware
* SDK: 27
* Release: 8.1.0
* Incremental: 1535356309

### Cause of error
```java
Exception in thread "AsyncTask #1": java.lang.RuntimeException: An error occurred while executing doInBackground()
	at android.os.AsyncTask$3.done([AsyncTask.java:353](http://asynctask.java:353/))
	at java.util.concurrent.FutureTask.finishCompletion([FutureTask.java:383](http://futuretask.java:383/))
	at java.util.concurrent.FutureTask.setException([FutureTask.java:252](http://futuretask.java:252/))
	at [java.util.concurrent.FutureTask.run](http://java.util.concurrent.futuretask.run/)([FutureTask.java:271](http://futuretask.java:271/))
	at java.util.concurrent.ThreadPoolExecutor.runWorker([ThreadPoolExecutor.java:1162](http://threadpoolexecutor.java:1162/))
	at java.util.concurrent.ThreadPoolExecutor$[Worker.run](http://worker.run/)([ThreadPoolExecutor.java:636](http://threadpoolexecutor.java:636/))
	at [java.lang.Thread.run](http://java.lang.thread.run/)([Thread.java:764](http://thread.java:764/))
Caused by: java.lang.NoSuchMethodError: No interface method toList()Ljava/util/List; in class Ljava/util/stream/Stream; or its super classes (declaration of '[java.util.stream.Stream](http://java.util.stream.stream/)' appears in /system/framework/core-oj.jar)
	at de.dennisguse.opentracks.Startup$$ExternalSyntheticApiModelOutline0.m(Unknown Source:0)
	at de.dennisguse.opentracks.util.ExportUtils.exportTrack([ExportUtils.java:56](http://exportutils.java:56/))
	at [de.dennisguse.opentracks.io](http://de.dennisguse.opentracks.io/).file.exporter.ExportService.onHandleWork([ExportService.java:55](http://exportservice.java:55/))
	at [androidx.core.app](http://androidx.core.app/).JobIntentService$CommandProcessor.doInBackground([JobIntentService.java:396](http://jobintentservice.java:396/))
	at [androidx.core.app](http://androidx.core.app/).JobIntentService$CommandProcessor.doInBackground([JobIntentService.java:387](http://jobintentservice.java:387/))
	at android.os.AsyncTask$[2.call](http://2.call/)([AsyncTask.java:333](http://asynctask.java:333/))
	at [java.util.concurrent.FutureTask.run](http://java.util.concurrent.futuretask.run/)([FutureTask.java:266](http://futuretask.java:266/))
	... 3 more
```

sgofferj, 2023-10-17:
Clarification: Using the app works just fine apparently but sometimes when I unlock my phone I get the Opentracks crash screen.

dennisguse, 2023-10-17:
Likely due to https://github.com/OpenTracksApp/OpenTracks/commit/8c8a7d9496f77b1d8616fa918b434b1a4e7f7796

Are you using post-workout exports?

sgofferj, 2023-10-17:
Yes, indeed I do. Auto-export to GPX.

dennisguse, 2023-10-17:
No worries ;)

Due to 8c8a7d9496f77b1d8616fa918b434b1a4e7f7796
Change reverted release as v4.9.3

closed 2023-10-17 by dennisguse