OpenTracksApp/OpenTracks #1738: Github crash report doesn't fill crash data in template

sgofferj, 2023-10-17:
**Describe the bug**
When OT crashes, there's a bug button which should open a bug report on Github. Well, it does but it leads to the bug template selection but when "bug" is selected, the information previously seen on the screen is not filled into the template.

**To Reproduce**
1. Crash the app
2. Click on the bug
3. Choose a template
4. See error

**Technical information**
### App information
* ID: de.dennisguse.opentracks.nightly
* Version: 5483 v4.9.2-17-g8f9ba046c

### Device information
* Brand: ulefone
* Device: Armor_3
* Model: Armor_3
* Id: O11019
* Product: Armor_3

### Firmware
* SDK: 27
* Release: 8.1.0
* Incremental: 1535356309

dennisguse, 2023-10-17:
Are you using the Github App or the Webpage?

@pstorch FYI

pstorch, 2023-10-17:
Hmm, maybe we have to add the issue template parameter to the report_issue_link like "&template=bug_report.md"

sgofferj, 2023-10-17:
@dennisguse I have the github app installed at least. I check on the next crash, what actually gets opened.

pstorch, 2023-10-17:
I cannot reproduce. When I send a bug report via browser (Firefox), I don't see the template selection and the issue contains all the info.
What is the difference?

sgofferj, 2023-10-18:
Checked today when it crashed again. It's opening the GitHub app, not GitHub in a browser.

dennisguse, 2023-10-18:
@sgofferj did you update to the newest nightly?

sgofferj, 2023-10-18:
I thought I did. Checking now.

sgofferj, 2023-10-20:
Well, now that I updated to latest nightly, OT is not crashing anymore, so I can't test the interaction with the Github app anymore...

dennisguse, 2023-10-21:
I just made OpenTracks crash and the Github app is opened, but requesting to select a template.
After selecting the template, just the template is shown (not the OpenTracks provided body).

pstorch, 2023-10-21:
I didn't had the Github App, only tested via browser. 
I now installed the Github App to test. When I provoke a crash the Github App opens with the issue without the template selection, but it misses the stacktrace in the issue description.

pstorch, 2023-10-24:
@dennisguse can you test with disabled Github App, so the browser opens the page?

dennisguse, 2024-12-28:
Just tested this:
* browser works
* github app doesn't

dennisguse, 2025-06-22:
Just tested this again. Github app now works.

closed 2025-06-22 by dennisguse