OpenTracksApp/OpenTracks #1741: Crash on Start Record

FHenry, 2023-10-17:
**Describe the bug**
Since last PalyStore Update the start record activity crash

**To Reproduce**
1. Activate GPS in the application
2. click on red button "Records tracks"
3. Application crash with this log

### App information
* ID: de.dennisguse.opentracks.playstore
* Version: 5465 v4.9.2-PlayStore

### Device information
* Brand: Xiaomi
* Device: tucana
* Model: Mi Note 10
* Id: QKQ1.190825.002
* Product: tucana_eea

### Firmware
* SDK: 29
* Release: 10
* Incremental: V12.0.1.0.QFDEUXM

### Cause of error
```java
Exception in thread "main": java.lang.NoSuchMethodError: No interface method toList()Ljava/util/List; in class Ljava/util/stream/Stream; or its super classes (declaration of 'java.util.stream.Stream' appears in /apex/com.android.runtime/javalib/core-oj.jar)
	at de.dennisguse.opentracks.Startup$$ExternalSyntheticApiModelOutline0.m(Unknown Source:0)
	at de.dennisguse.opentracks.settings.PreferencesUtils.getMultiTypedArray(PreferencesUtils.java:732)
	at de.dennisguse.opentracks.settings.PreferencesUtils.buildDefaultFields(PreferencesUtils.java:737)
	at de.dennisguse.opentracks.settings.PreferencesUtils.buildDefaultLayout(PreferencesUtils.java:743)
	at de.dennisguse.opentracks.settings.PreferencesUtils.getCustomLayout(PreferencesUtils.java:754)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.lambda$new$0(StatisticsRecordingFragment.java:66)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.$r8$lambda$tMysS2hXjodPYdIchQvJybasmtA(Unknown Source:0)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment$$ExternalSyntheticLambda0.onSharedPreferenceChanged(Unknown Source:2)
	at de.dennisguse.opentracks.settings.PreferencesUtils.registerOnSharedPreferenceChangeListener(PreferencesUtils.java:86)
	at de.dennisguse.opentracks.fragments.StatisticsRecordingFragment.onResume(StatisticsRecordingFragment.java:89)
	at androidx.fragment.app.Fragment.performResume(Fragment.java:3039)
	at androidx.fragment.app.FragmentStateManager.resume(FragmentStateManager.java:607)
	at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:306)
	at androidx.fragment.app.FragmentManager.executeOpsTogether(FragmentManager.java:2189)
	at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute(FragmentManager.java:2100)
	at androidx.fragment.app.FragmentManager.execSingleAction(FragmentManager.java:1971)
	at androidx.fragment.app.BackStackRecord.commitNow(BackStackRecord.java:305)
	at androidx.viewpager2.adapter.FragmentStateAdapter$FragmentMaxLifecycleEnforcer.updateFragmentMaxLifecycle(FragmentStateAdapter.java:726)
	at androidx.viewpager2.adapter.FragmentStateAdapter.placeFragmentInViewHolder(FragmentStateAdapter.java:342)
	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:276)
	at androidx.viewpager2.adapter.FragmentStateAdapter.onViewAttachedToWindow(FragmentStateAdapter.java:67)
	at androidx.recyclerview.widget.RecyclerView.dispatchChildAttached(RecyclerView.java:7556)
	at androidx.recyclerview.widget.RecyclerView$5.addView(RecyclerView.java:860)
	at androidx.recyclerview.widget.ChildHelper.addView(ChildHelper.java:107)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addViewInt(RecyclerView.java:8601)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addView(RecyclerView.java:8559)
	at androidx.recyclerview.widget.RecyclerView$LayoutManager.addView(RecyclerView.java:8547)
	at androidx.recyclerview.widget.LinearLayoutManager.layoutChunk(LinearLayoutManager.java:1641)
	at androidx.recyclerview.widget.LinearLayoutManager.fill(LinearLayoutManager.java:1587)
	at androidx.recyclerview.widget.LinearLayoutManager.onLayoutChildren(LinearLayoutManager.java:665)
	at androidx.recyclerview.widget.RecyclerView.dispatchLayoutStep2(RecyclerView.java:4134)
	at androidx.recyclerview.widget.RecyclerView.dispatchLayout(RecyclerView.java:3851)
	at androidx.recyclerview.widget.RecyclerView.onLayout(RecyclerView.java:4404)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at androidx.viewpager2.widget.ViewPager2.onLayout(ViewPager2.java:527)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.layoutChild(CoordinatorLayout.java:1213)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.onLayoutChild(CoordinatorLayout.java:899)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.onLayout(CoordinatorLayout.java:919)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1829)
	at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1673)
	at android.widget.LinearLayout.onLayout(LinearLayout.java:1582)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.layoutChild(CoordinatorLayout.java:1213)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.onLayoutChild(CoordinatorLayout.java:899)
	at androidx.coordinatorlayout.widget.CoordinatorLayout.onLayout(CoordinatorLayout.java:919)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:332)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:270)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:332)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:270)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:332)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:270)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at android.widget.LinearLayout.setChildFrame(LinearLayout.java:1829)
	at android.widget.LinearLayout.layoutVertical(LinearLayout.java:1673)
	at android.widget.LinearLayout.onLayout(LinearLayout.java:1582)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at android.widget.FrameLayout.layoutChildren(FrameLayout.java:332)
	at android.widget.FrameLayout.onLayout(FrameLayout.java:270)
	at com.android.internal.policy.DecorView.onLayout(DecorView.java:812)
	at android.view.View.layout(View.java:22153)
	at android.view.ViewGroup.layout(ViewGroup.java:6325)
	at android.view.ViewRootImpl.performLayout(ViewRootImpl.java:3191)
	at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2701)
	at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1819)
	at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7781)
	at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1031)
	at android.view.Choreographer.doCallbacks(Choreographer.java:854)
	at android.view.Choreographer.doFrame(Choreographer.java:789)
	at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1016)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:224)
	at android.app.ActivityThread.main(ActivityThread.java:7584)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
```

dennisguse, 2023-10-17:
Release submitted to PlayStore.
Hopefully they process it faster than last time (5 days).
F-Droid: as usual 3-5 days.

closed 2023-10-17 by dennisguse

FHenry, 2023-10-18:
Release on playStore.
Test and Fix !
Thank's