OpenTracksApp/OpenTracks #1746: Update to Android 14 breaks when starting run.

rutrum, 2023-10-25:
### App information
* ID: de.dennisguse.opentracks
* Version: 5491 v4.9.3

### Device information
* Brand: google
* Device: bluejay
* Model: Pixel 6a
* Id: UP1A.231005.007.A1
* Product: bluejay

### Firmware
* SDK: 34
* Release: 14
* Incremental: 10762838

### Cause of error
```java
Exception in thread "main": java.lang.SecurityException: Starting FGS with type connectedDevice callerApp=ProcessRecord{e3a1b7a 5478:de.dennisguse.opentracks/u0a222} targetSDK=34 requires permissions: all of the permissions allOf=true [android.permission.FOREGROUND_SERVICE_CONNECTED_DEVICE] any of the permissions allOf=false [android.permission.BLUETOOTH_ADVERTISE, android.permission.BLUETOOTH_CONNECT, android.permission.BLUETOOTH_SCAN, android.permission.CHANGE_NETWORK_STATE, android.permission.CHANGE_WIFI_STATE, android.permission.CHANGE_WIFI_MULTICAST_STATE, android.permission.NFC, android.permission.TRANSMIT_IR, android.permission.UWB_RANGING, USB Device, USB Accessory] 
	at android.os.Parcel.createExceptionOrNull(Parcel.java:3057)
	at android.os.Parcel.createException(Parcel.java:3041)
	at android.os.Parcel.readException(Parcel.java:3024)
	at android.os.Parcel.readException(Parcel.java:2966)
	at android.app.IActivityManager$Stub$Proxy.setServiceForeground(IActivityManager.java:6761)
	at android.app.Service.startForeground(Service.java:862)
	at de.dennisguse.opentracks.Startup$$ExternalSyntheticApiModelOutline0.m(Unknown Source:0)
	at de.dennisguse.opentracks.services.TrackRecordingService.startSensors(TrackRecordingService.java:202)
	at de.dennisguse.opentracks.services.TrackRecordingService.startRecording(TrackRecordingService.java:187)
	at de.dennisguse.opentracks.services.TrackRecordingService.startNewTrack(TrackRecordingService.java:168)
	at de.dennisguse.opentracks.TrackListActivity.lambda$onCreate$5(TrackListActivity.java:187)
	at de.dennisguse.opentracks.TrackListActivity.$r8$lambda$9VATjDTlXeRwAYrK9Lb8GzlwA8U(Unknown Source:0)
	at de.dennisguse.opentracks.TrackListActivity$$ExternalSyntheticLambda8.onConnected(Unknown Source:2)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.setTrackRecordingService(TrackRecordingServiceConnection.java:147)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.onServiceConnected(TrackRecordingServiceConnection.java:162)
	at android.app.LoadedApk$ServiceDispatcher.doConnected(LoadedApk.java:2198)
	at android.app.LoadedApk$ServiceDispatcher$RunConnection.run(LoadedApk.java:2231)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:294)
	at android.app.ActivityThread.main(ActivityThread.java:8177)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
Caused by: android.os.RemoteException: Remote stack trace:
	at com.android.server.am.ActiveServices.validateForegroundServiceType(ActiveServices.java:2611)
	at com.android.server.am.ActiveServices.setServiceForegroundInnerLocked(ActiveServices.java:2322)
	at com.android.server.am.ActiveServices.setServiceForegroundLocked(ActiveServices.java:1679)
	at com.android.server.am.ActivityManagerService.setServiceForeground(ActivityManagerService.java:13265)
	at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:3385)

```

dennisguse, 2023-10-25:
I assume you upgraded from Android 13 to 14, right?
Did OpenTracks requests any missing permissions when you started it for the first time on Android 14?

Can you check that all permissions are (still) granted?

dennisguse, 2023-10-25:
PS/ #1655 reported a similar (or even same) problem.

rutrum, 2023-10-25:
Ah, Im sorry. It did not reprompt me with a request for more permissions, but that was the solution: enabling access to nearby devices. Thank you for your help.

closed 2023-10-25 by rutrum

dennisguse, 2023-10-26:
Good to know. 
I am not sure if we have to do (or can do) something about this. 
Hope that it is not happening too often.