OpenTracksApp/OpenTracks #1748: Idle voice announcements are played even though they are deactivated in settings

bneuburg, 2023-10-27:
**Describe the bug**

Since at least 2023/10/24 OpenTracks started playing idle announcements ("Becoming idle") on my bluetooth headset, even though "Idle" is deactivated under 'Settings > Statistics Announcements > Idle Announcements' 

When turning on 'Settings > Statistics Announcements >  Use the device's speaker'  and disconnecting the bluetooth headset idle voice announcements are also played over the normal speaker. 

When deactivating this option and disconnecting the bluetooth headset, no voice announcements are played (as expected):
`10-27 11:13:31.146 30425 30425 I VoiceAnnouncementManager: No voice announcement on device speaker.`

I have also activated `Idle announcements` and deactivated it again. Voice notifications play in both cases.

This change in behavior is probaly related to #1707 

**To Reproduce**
1. Go to  'Settings > Statistics Announcements > Use the device's speaker' 
2. Activate this in order to reproduce without headset
3. Scroll to 'Idle Announcements'
4. Make sure 'Idle' is turned off
5. Go to 'Settings > GPS'
6. Set 'GPS accuracy' to '10 m' and 'Recording distance interval' to '1 m' to trigger idle detection more often in a closed space
7. Run `adb logcat | grep Becoming\ idle` and each time a new idle detection happens the voice announcement is played

** Further Information **
adb logs report the following error regarding `VoiceAnnouncmentManager`, not sure if this matters:

`0-27 10:44:46.531 30425 30425 E PreferenceGroup: Found duplicated key: "voiceAnnounceTotalDistance". This can cause unintended behaviour, please use unique keys for every preference.`

**Technical information**
 - Device: Samsung S21 FE
 - OS: Latest stock firmware, Android 13
 - OpenTracks version: v4.9.3 from Fdroid, version code 5491

dennisguse, 2023-10-27:
Seems I messed up the preference config.
Could you fix that?

bneuburg, 2023-10-31:
@dennisguse I can try to fix it if you can give me a pointer where to do that, I haven't tinkered with Android apps yet.

After looking around I suspect the problem is in [settings_announcments.xml](https://github.com/OpenTracksApp/OpenTracks/blob/e9e9bb387420baffbdde93d0ddcc9a8da7bddec3/src/main/res/xml/settings_announcements.xml#L20)
```
    <PreferenceCategory android:title="@string/settings_announcements_idle_category_title">
        <SwitchPreferenceCompat
            android:defaultValue="@bool/voice_announce_total_distance_default"
            android:key="@string/voice_announce_total_distance_key"
            android:title="@string/settings_announcements_idle_title" />
    </PreferenceCategory>
```

That should probably be

```
    <PreferenceCategory android:title="@string/settings_announcements_idle_category_title">
        <SwitchPreferenceCompat
            android:defaultValue="@bool/voice_announce_idle_default"
            android:key="@string/voice_announce_idle_key"
            android:title="@string/settings_announcements_idle_title" />
    </PreferenceCategory>
```

dennisguse, 2023-10-31:
@bneuburg that seems to be the solution.
Can you create a PR for this?

bneuburg, 2023-10-31:
@dennisguse Done

dennisguse, 2023-11-01:
Closed by #1751.

@bneuburg Thanks :D

closed 2023-11-01 by dennisguse