OpenTracksApp/OpenTracks #1749: App crashes with permission errors

asherp, 2023-10-28:
### App information
* ID: de.dennisguse.opentracks
* Version: 5491 v4.9.3

### Device information
* Brand: google
* Device: oriole
* Model: Pixel 6
* Id: UP1A.231005.007
* Product: oriole

### Firmware
* SDK: 34
* Release: 14
* Incremental: 2023102300

### Cause of error
```java
Exception in thread "main": java.lang.SecurityException: Starting FGS with type connectedDevice callerApp=ProcessRecord{7c98841 32708:de.dennisguse.opentracks/u0a154} targetSDK=34 requires permissions: all of the permissions allOf=true [android.permission.FOREGROUND_SERVICE_CONNECTED_DEVICE] any of the permissions allOf=false [android.permission.BLUETOOTH_ADVERTISE, android.permission.BLUETOOTH_CONNECT, android.permission.BLUETOOTH_SCAN, android.permission.CHANGE_NETWORK_STATE, android.permission.CHANGE_WIFI_STATE, android.permission.CHANGE_WIFI_MULTICAST_STATE, android.permission.NFC, android.permission.TRANSMIT_IR, android.permission.UWB_RANGING, USB Device, USB Accessory] 
	at android.os.Parcel.createExceptionOrNull(Parcel.java:3066)
	at android.os.Parcel.createException(Parcel.java:3050)
	at android.os.Parcel.readException(Parcel.java:3026)
	at android.os.Parcel.readException(Parcel.java:2968)
	at android.app.IActivityManager$Stub$Proxy.setServiceForeground(IActivityManager.java:6776)
	at android.app.Servic

asherp, 2023-10-28:
I was able to get it working by

1. Exporting tracks
2. Uninstall app
3. Reinstall app
4. Allow permissions when prompted

dennisguse, 2023-10-28:
Mhh.
That's report number three. 

Can you try to reproduce this?
My guess is that the Bluetooth permission now implicitly became mandatory. 
Can you check if you remove this in Android settings for OpenTracks (ask or deny). 
... And see what happens.

fabibau, 2023-10-28:
Same issue here. After allowing nearby devices in settings it worked again.

dennisguse, 2023-10-28:
@fabibau can you make a screenshot of the UI where you allowed nearby devices?
I only have an Android 13 at hand...

fabibau, 2023-10-28:
[image]
It is the second permission "Geräte in der Nähe".

dennisguse, 2023-10-28:
I guess, I understood the problem: Bluetooth permissions are now always mandatory. The same applies to GPS. 

OpenTracks cannot start it's foreground service without this anymore. Until Android 13 those permissions are optional.

dennisguse, 2023-10-29:
@fabibau @asherp Could you test this? See #1750 

@pstorch you as well?

pstorch, 2023-10-29:
I'm still on Android 13.

fabibau, 2023-10-29:
@dennisguse With the nightly build from your PR the app does not crash any longer if you deny the nearby devices permission. It also asks for granting it at startup and seems to work properly.

closed 2023-10-29 by dennisguse

dennisguse, 2023-10-30:
@fabibau thanks for testing; released as v4.9.5

PR #1750 (2023-10-29): Bugfix (Android14): for recording Location/GPS and Nearby Devices/Blu…
…etooth are mandatory due to Android.

Fixes #1749.
commit cb48d7f7 (2023-10-29): Bugfix (Android14): for recording Location/GPS and Nearby Devices/Bluetooth are mandatory due to Android.

Fixes #1749.