OpenTracksApp/OpenTracks #1756: Since V4.90 movement detection inactive

lsuhrbier, 2023-11-05:
Since the update to V4.90 the app can not distinguish anymore between movement time and overall time. 

After having started a track and going to a bike tour, both times are always the same. Before, the movement time stopped automatically on stops. I configured an idle time (which is only possible for GPS, but not for bluetooth sensors), but this appears not to have any impact. I am using a bluetooth speed sensor.

Nevertheless, the app appears to track it. When I change to OSM Dashboard to watch my track, I can see a lot of coffee cups there, but no time differences between movement and overall time.

dennisguse, 2023-11-06:
The idle time should apply to all sensors (it is still in the wrong place in the settings though).
Can you check if idle gets detected by enabling the voice announcements for idle?

lsuhrbier, 2023-11-06:
I checked it today and turned most voice announcements on. But I got no idle announcements, just average speed, heart frequency etc.

But regarding heart frequency, may be I have an idea. I always use the app with bluetooth heart frequency sensor.  As you mentioned idle time applies to all sensors, may be idle detection also does in the new release ? And the heart frequency sensor does not stop sending data when movement has stopped... So, may be that's the problem ?

dennisguse, 2023-11-06:
Actually, that might be the problem.
Can you try without the heartrate sensor?

lsuhrbier, 2023-11-08:
I tried without heartrate sensor today. But no change. I don't get any idle-announcements and both times are identical during the whole track.

Mats391, 2023-11-09:
I also noticed this issue. When riding my bike I use Wahoo speed+candence sensors as well as heartrate sensor. 
Right now sitting at my desk and it still thinks I am moving. Even when removing all the bluetooth sensors in the settings, it still detects as moving. The only sensor I was not able to deactivate is the barometric sensor (Pixel 6a). Maybe that one can trigger this issue?

Mats391, 2023-11-09:
I think it does. I just called it wrong. That article lists "Barometer" under Sensoren and gsmarena also lists it for Pixel 5 and Pixel 5A 5g 
https://www.gsmarena.com/google_pixel_5-10386.php 
https://www.gsmarena.com/google_pixel_5a_5g-11059.php

lsuhrbier, 2023-11-09:
Ok, checked it with the Barometer App from Google Play Store. It appears to have a barometer sensor, as it shows pressure without GPS enabled/allowed.

dennisguse, 2023-11-09:
You might be right; the relevant code is in TrakRecordingManager.onNewTrackPoint(). 

There is a timer that is started and if it triggers idle is assumed. 
Maybe I missed a case for using other sensors.

Mats391, 2023-11-25:
I did some more testing and in addition to removing all Bluetooth sensors, I used the developer options to disable all internal sensors. Even with no sensors at all, it still does not detect idle.

dennisguse, 2023-11-25:
Then my next guess is some kind of powersaving.
Can you check what happens if the display is on?

And do you use the idle announcements?

Mats391, 2023-11-28:
I noticed a flaw in my testing. If you never moved and everything stays at 0, it never switches to idle. I redid the testing by first moving a bit and then sitting down. That way the idle got detected correctly and moving time did no longer climb. With this setting I tested so far:
* No Bluetooth Sensors + Sensors off: Detection works
* No Bluetooth Sensors + Sensors on: Detection works. Even shaking phone to trigger barometer bug does not restart moving time
* Added heart rate sensor: Detection works

This makes it look like it might be related to speed or cadence sensor. I will test those later today when I am back on the bike.

Mats391, 2023-11-28:
Did testing on the bike with the different sensors:
* GPS + Heart rate: Idle works
* GPS + Heart rate + Cadance: Idle works
* Speed + Heart rate + Cadance: Idle does not work
* Speed + Heart rate: Idle does not work
* Only Speed: Idle does not work
* Only Speed + Sensors off: Idle does not work
So in my case the problem seems to be coming from the speed sensor.

dennisguse, 2023-11-28:
Nice digging!
The speed sensor actually provides distance, so we need to check that we handle it correctly if it doesnt report any movement.

lsuhrbier, 2023-12-27:
I checked the latest version today(i.e. 5563, v4.9.8). It looks like idle recognition works now as expected including idle-announcements. 

But, there are other problems now. 
- Cadence display shows drop outs (just shows '-' like as it is not connected) in irregular intervals of 1-5 sec.
- After a couple of idle recognitions, or whenever stopping tracking manually the application crashes. 
- It is possible to continue a track after a crash, but then neither speed nor cadence sensors are reconnecting successfully. Speed is then tracked using GPS, cadence is not tracked, heart frequency continues to be tracked as usual.

dennisguse, 2023-12-27:
@lsuhrbier Thanks for the report.
Please upgrade to v4.9.9 (released today) as it contains some fixes against crashes.
The sensor subsystem was created twice and thus not correctly shutdown.
This might also result in the issue of reconnecting not working.

The cadence problem is tracked here: #1790
As I have no sensor at hand, I have hard to time to check this easily.
I hope I have some time next week (and a sensor).

Mats391, 2023-12-28:
I can confirm that with 4.9.9-16 nightly the movement detection worked for me.

lsuhrbier, 2024-02-07:
Idle recognition does not work anymore. 

Even though, the voice says "becoming idle", boths clocks are showing the same time, the average speed continuously goes down when having stopped and the speed and cadence display are probably show the last measured values (speed or frequency) before movement has been stopped.

dennisguse, 2024-04-01:
@lsuhrbier I can confirm this behavior.
It is in TrackRecordingManager:203; it used to store sensor data if these do not provide a location/or distance.
This happens every 10s (hardcoded), so idle would not be triggered at all if a sensor like heartrate is connected.

We have the following options:
1. remove functionality to store sensor data if no movement is happening
2. restrict this functionality to only store while moving aka not idle (interesting for cadence/power/barometer)
3. keep functionality but actually fix moving time (interesting mainly for heartrate to measure recovery behavior)

@lsuhrbier What do you think?

lsuhrbier, 2024-04-03:
Ok, please keep digging !!!
Regarding the options, I just can say that i am using BT-heartrate, cadence and speed sensors when using it to track my bike tours. 

Option 1 and 2 appear to me to finally offering the same functionality, i.e. not saving any sensor data when not moving. Both options are not offering the expected behaviour, but would be better than the current situation. 

I do not really understand, what's option 3's impact is (fix moving time?). 

Hope you can solve the problem somehow.

G-i-o, 2024-05-09:
Hi. FYI, I have the same issue on Pixel 3a, using latest version (v.4.12.3 updated today....unfortunately :-( ). No BT devices attached..
Thank You.

keraba, 2024-05-10:
Same issue here, on a Moto Razr Plus, no bluetooth anything. I hear "Becoming idle" quite often, but the "total movement time" is identical to actual time. April 13 worked; 20th didn't.

Maybe we should make a separate issue?

KN4CK3R, 2024-05-18:
I did a downgrade to 4.11.3 from 4.12.3. This was the last version with working idle detection (for me).

CastelJeremy, 2024-05-25:
I did some tests on different devices (Xiaomi Redmi A2 and Fairphone 5).
On both of them I could get different "moving time" and "global time" with v4.12.1 and prior, but not with v4.12.2 or v4.12.3.
I didn't try to get the "Becoming idle" message though.

I'm guessing v4.12.2 probably broke something.

Random90, 2024-05-25:
I have the same issue and it appeared around May 2. Before that moving time was shorter than total time like it should.
Version: v4.12.3
OS: GrapheneOS

vlmendz, 2024-05-26:
For me the idle detection functionality apparently stopped working in an update between April 19th (last recording with time and speed in motion ok) and April 21.

Checking the recorded tracks I have noticed that idle detection is actually working (only are recorded positions when there is movement), but for some reason, the same values of time and speed are shown, both in movement and total.

OMTDesign, 2024-06-10:
In my case Opentracks does not report a zero value from my speed sensor, instead it holds the last speed value from the speed sensor, which tricks the app into thinking that there is movement. Unparing the speed sensor restores movement detection, as mentioned by Mats391 in his post.

christfried, 2024-06-13:
For me movement detection stopped working between the 28th of December and the 15th of January. Since January there was also no tracking of altitude anymore. Sadly I cannot remember changing any settings that may have led to that behavior.
I never used any external sensors though.
For the last two runs I changed the Barometer from "none" to "internal sensor" but it didn't help regarding the movement detection.

closed 2024-07-05 by dennisguse

dennisguse, 2024-07-06:
Took quite a while, but this got finally fixed.
Released in https://github.com/OpenTracksApp/OpenTracks/releases/tag/v4.12.5

dennisguse, 2024-07-31:
@lsuhrbier great!
Can you create a new issue for the still existing problem?

commit 36f41aa0 (2024-07-04): Bugfix: moving time was not computed.

Introduced in 6be9d2477cfa9f3c6a3d800b291d68717f6c49ba (first released in v4.9.0)

Fixes #1756.
commit fa820102 (2024-07-05): Bugfix: moving time was not computed.

Introduced in 6be9d2477cfa9f3c6a3d800b291d68717f6c49ba (first released in v4.9.0)

Fixes #1756.
commit 85aac6fa (2024-07-05): Bugfix: moving time was not computed.

Introduced in 6be9d2477cfa9f3c6a3d800b291d68717f6c49ba (first released in v4.9.0)

Fixes #1756.
commit 5f23c530 (2024-07-05): Bugfix: moving time was not computed.

Introduced in 6be9d2477cfa9f3c6a3d800b291d68717f6c49ba (first released in v4.9.0)

Fixes #1756.