OpenTracksApp/OpenTracks #1760: SRTM correction for GPX

DavidHenryThoreau, 2023-11-10:
Add SRTM correction like OsmAnd~ app (a button to corect SRTM after gpx record to correct elevation en km)

 
Here's documentation about it : 
https://www.osmand.net/docs/technical/algorithms/calculate-uphill-slope#altitude-srtm-correction

dennisguse, 2023-11-11:
OpenTracks stores elevation as WGS84, but shows it as EGM2008 (using data included in the app).
Why do you need SRTM?

DavidHenryThoreau, 2023-11-11:
Because there's lot of difference between the GPX about elevation gain and the elevation gain in the reality, that's why i'm using SRTM in all tracking application.

DavidHenryThoreau, 2023-11-11:
Because there's lot of differences between the GPX record elevation gain and the elevation gain in the reality, that's why i'm using SRTM in all tracking application after I've recorded the GPX file at the moment i'm using OsmAnd~

luzpaz, 2023-12-05:
what are the differences ?

DavidHenryThoreau, 2023-12-05:
Lot's of differences : elevation gain