OpenTracksApp/OpenTracks #1761: Ability to directly upload to Fitness Trackers (Specifically, custom FitTrackee)

Firesphere, 2023-11-15:
**Is your feature request related to a problem? Please describe.**

When I want to put my rides in to [FitTrackee](https://github.com/SamR1/FitTrackee), I need a GPX file from somewhere.

This process is quite cumbersome.

**Describe the solution you'd like**

Either:
- Ability to upload directly from the app to FitTrackee
- Easier way to use my phone's browser to get to the export files in GPX format

**Describe alternatives you've considered**

Use SuperCycle, because it does have GPX

dennisguse, 2023-11-15:
Current options to export (to Android device file system):
* Automatic post-recording export
* Share track functionality
* Export via settings (all in one or all individually)

How AAT is doing by providing an Android data source would be interesting. 

Anyhow, the upload functionality must be done by another application like Nextcloud or Syncthing.

Firesphere, 2023-11-15:
Ahhhh, the automatic post-recording would solve about half of my problem 👍

closed 2024-03-01 by dennisguse