OpenTracksApp/OpenTracks #1765: Add icons to some activities types

ScriptForAll, 2023-11-19:
**Is your feature request related to a problem? Please describe.**
I use OpenTracks in order to keep track of my mobility patterns. Whenever I start scrolling to find trips that use a given mode of transportation in some cases this can be more difficult. 
Such cases include "driving bus" that shares the same icon as "driving car", and "train" that only has the general icon for activities.

**Describe the solution you'd like**
Change the icons of "driving bus" and "train" so that the user can distinguish these particular activities from the rest. 

**Describe alternatives you've considered**
NA

**Additional context**
NA

dennisguse, 2023-11-20:
@ScriptForAll if you have an icon in mind, please provide it. It is relatively easy to add.

ScriptForAll, 2023-11-22:
Where do the icons that are being used in the app come from?

dennisguse, 2023-11-22:
Actually I am not sure; most of them were part of MyTracks and others were contributed.
The only thing that is important it that those icons do not violate any copyright.

closed 2024-12-19 by dennisguse