OpenTracksApp/OpenTracks #1768: Crash on track selection followed by "Edit"

dhlocker, 2023-11-23:
**Describe the bug**
Editing one track from track list causes crash

**To Reproduce**
1. Open app
2. Long press a track to select it (note background changes)
3. click "triple dots" menu and click "Edit"
4. Observe "OpenTracks crashed"...

**Expected Behaviour**
1. Edit works when invoked this way: Open app
2. click to open one track
3. click "triple dots"
4. click Edit
5. Edit Name/Activity type/Description and Save

**Technical information**
 ### App information
* ID: de.dennisguse.opentracks
* Version: 5399 v4.8.4

### Device information
* Brand: samsung
* Device: r0q
* Model: SM-S901U
* Id: TP1A.220624.014
* Product: r0qsqw

### Firmware
* SDK: 33
* Release: 13
* Incremental: S901USQS3CWI1

### Cause of error
```java
Exception in thread "main": java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.ui.TrackListAdapter$ViewHolder.setSelected(boolean)' on a null object reference
at de.dennisguse.opentracks.ui.TrackListAdapter.setAllSelected(TrackListAdapter.java:135)
at de.dennisguse.opentracks.ui.TrackListAdapter.onDestroyActionMode(TrackListAdapter.java:126)
at androidx.appcompat.app.AppCompatDelegateImpl$ActionModeCallbackWrapperV9.onDestroyActionMode(AppCompatDelegateImpl.java:3027)
at androidx.appcompat.view.StandaloneActionMode.finish(StandaloneActionMode.java:110)
at de.dennisguse.opentracks.ui.TrackListAdapter.onActionItemClicked(TrackListAdapter.java:117)
at androidx.appcompat.app.AppCompatDelegateImpl$ActionModeCallbackWrapperV9.onActionItemClicked(AppCompatDelegateImpl.java:3022)
at androidx.appcompat.view.StandaloneActionMode.onMenuItemSelected(StandaloneActionMode.java:140)
at androidx.appcompat.view.menu.MenuBuilder.dispatchMenuItemSelected(MenuBuilder.java:836)
at androidx.appcompat.view.menu.MenuItemImpl.invoke(MenuItemImpl.java:159)
at androidx.appcompat.view.menu.MenuBuilder.performItemAction(MenuBuilder.java:987)
at androidx.appcompat.view.menu.MenuPopup.onItemClick(MenuPopup.java:128)
at android.widget.AdapterView.performItemClick(AdapterView.java:376)
at android.widget.AbsListView.performItemClick(AbsListView.java:1287)
at android.widget.AbsListView$PerformClick.run(AbsListView.java:3563)
at android.widget.AbsListView$3.run(AbsListView.java:4716)
at android.os.Handler.handleCallback(Handler.java:942)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loopOnce(Looper.java:226)
at android.os.Looper.loop(Looper.java:313)
at android.app.ActivityThread.main(ActivityThread.java:8762)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:604)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
```

dhlocker, 2023-11-23:
Latest version (5527 v.4.9.5) works fine. F-Droid reported all up-to-date, but after "refreshing repos" OpenTracks showed an update available. Updated and all is OK.

closed 2023-11-23 by dhlocker

dennisguse, 2023-11-23:
@dhlocker thanks for the detailed report :)