OpenTracksApp/OpenTracks #1769: Broken .gpx export

pstorch, 2023-11-26:
I exported again all my activities from OpenTracks to .gpx. Some files have errors like this, see the broken element `ension>`:

```
<trkpt lat="50.xxxxxx" lon="9.yyyyyy">
<ele>427</ele>
<time>2020-06-11T18:16:43Z</time>
<extensions><gpxtpx:TrackPointExtension>
<gpxtpx:speed>3.05</gpxtpx:speed>
</gpxtpx:TrackPointExtension></extensions>
</trkpt>
<trkpt lat="50.xxxxxx" lon="9.yyyyyy">
<ele>430</ele>
<time>2020-06-11T18:16:48Z</time>
<extensions><gpxtpx:TrackPointExtension>
<gpxtpx:speed>1.75</gpxtpx:speed>
</gpxtpx:TrackPointExtension></extensions>
</trkpt>
</trkseg>
</trk>
</gpx>
ension>
<gpxtpx:speed>1.75</gpxtpx:speed>
</gpxtpx:TrackPointExtension></extensions>
</trkpt>
</trkseg>
</trk>
</gpx>

```

OpenTracks nightly v4.9.5-15-g8b7063d18

pstorch, 2023-11-26:
Some more hints:
I exported 250 tracks, from 2019-01-29 until 2023-11-19. 
28 files had this kind of error. They are all in the timeframe of 2020-02-19 until 2020-07-04. Probably there is something wrong in the data of this timeframe.

pstorch, 2023-11-26:
Found the cause: in this timeframe all tracks were doubled in the database. Probably due to an export import test without the duplicate detection.

dennisguse, 2023-11-26:
Some kind of race condition? Or are we reusing the Outputstream?

pstorch, 2023-11-26:
Don't know yet. On the first glance the code looks ok.

commit dfd5b560 (2023-11-28): fix #1769 export overwrite existing files
commit 87d7ebcb (2023-11-28): fix #1769 export overwrite existing files
closed 2023-11-28 by dennisguse