OpenTracksApp/OpenTracks #1770: Exported data isn't accurate.

Estebiu, 2023-11-27:
I recently switched phones, and found that exporting my workouts from the old one to the new one resulted in some slightly incorrect data.

**To Reproduce**
1. Export Workouts (tried exporting as kml and gpx, as a single file or as multiple files,no difference)
2. Open statistics; 
3. See difference between in the distance and time fields between data before and after exporting
Old phone: 
[image]
New Phone: 
[image]

How a track's info looks:
Old phone:
[image]
New phone:
[image]

**Technical information**
 - Device: Old phone: Redmi Note 5 Pro; New Phone: Redmi Note 12
 - OS: Old phone: Lineage 20; New phone: Miui14
 - OpenTracks version: Both 4.9.5

dennisguse, 2023-11-27:
That is kind of expected. 
Some time ago the computation of the track statistics was changed. 

However, updating the existing statistics is relatively complicated and we skipped this.  
And on the re-import, the new algorithm is used. 

PS might even be more than one change.

dennisguse, 2023-11-27:
IIRC 
* we didn't count some distance between certain points (no clue actually why)
* speed is now derived from traveled distance rather from GPS
... and there may be some more.

It is really tricky to keep all these parts identical and this is often not worth the effort (or rather everything else is more important).

PS/ nicely written bug report :+1: 
PPS/ quite some traveling with OpenTracks :sunflower:

Estebiu, 2023-11-28:
I don't mind the slight difference in the traveled kms, but can't something be done with the total/moving time? After importing the files the "paused" time converts to being "moving" time, which messes up also the "average speed" statistic.

dennisguse, 2023-11-28:
Mhh. There was another change: how idle is handled.
Now, idle events are stored as an extra point (using a timer) instead of being decided by if the speed is below a certain threshold.
IDLE: If no movement happened within a defined time frame.

I am not sure, but there should be some code for importing that also handles IF no idle points are present.

Estebiu, 2023-11-28:
That would be perfect :D 
Thanks for reopening.

dennisguse, 2023-11-28:
Is guess the relevant code is in the TrackImporter.

voklav, 2024-02-15:
Well,
even in with latest version ...  Recorded. -> Exported -> Deleted -> Imported in the same phone that was recorded ... and moving time and total time are the same.

dennisguse, 2024-02-16:
@voklav did this happen with a recording done with OpenTracks v4.11.X?
And if not can you try it again?

voklav, 2024-03-17:
> @voklav did this happen with a recording done with OpenTracks v4.11.X? And if not can you try it again?

I am extremely sorry for replying so late.
But there are no problems with the new versions.
New to new - everything is fine.

closed 2024-12-19 by dennisguse