OpenTracksApp/OpenTracks #1778: export route with LineString

chclxds, 2023-12-07:
Hello

It would be a feature, or an option if possible.
The application is excellent, but I would like it if when exporting in .kml format, the entire route traveled was treated as a line (LineString). If there is such an option, where do I activate it?

Q: Why would this be useful?
R: I work with a provider, and the line option would perfectly portray the route of the fiber, making mapping easier. Along with the points.

dennisguse, 2023-12-08:
This is not implemented.
It shouldn't be be too difficult to build.
The relevant code is in KMLTrackExporter.

chclxds, 2023-12-08:
I see.
In that case, there is a possibility that this feature is being developed.

dennisguse, 2023-12-27:
I was checking the provided documentation (finally got some spare time again).
Bad news: the LineString object only is a list of coordinates (lat, lng, elevation) as a string.
There is no way to add any sensor data or timestamps.
So, this would mean exporting the LineString would have to be done in addition and thus significantly increasing the file size without actually adding new data.
That's just not a good idea.

@chclxds If you really need these LineString objects, I would recommend to create a script that converts the existing `<when><coord>` combination into a LineString object in post-processing step.

closed 2023-12-27 by dennisguse