OpenTracksApp/OpenTracks #1779: Exception on Save

evanpeterjones, 2023-12-08:
**Describe the bug**
Exception occurs while saving new recording.

**To Reproduce**
1. Start recording 
2. Stop recording 
3. See error

If applicable:
* add screenshots
*
[image]

**Technical information**
 - Device: Pixel 6
 - OS: Android 13
 - OpenTracks version: v4.9.6

dennisguse, 2023-12-08:
Always or sometimes?

Mats391, 2023-12-08:
I can confirm that I get the same crash every time I stop recording. Is there a way to access the logs of older crashes? Yesterday I also got some crashes while recording with bluetooth sensors connected.

dennisguse, 2023-12-08:
Not like this. 
You can Android studio to access logcat.

dennisguse, 2023-12-09:
I added a quick fix; can you check the nightly?

dennisguse, 2023-12-09:
Added quick fix: https://github.com/OpenTracksApp/OpenTracks/commit/24212a226ec07765955b9cfaf925382daad2e413

Mats391, 2023-12-09:
I tested the nightly and the barometer crash is gone for me. I created a new issue for the other bluetooth related crashes I mentioned earlier: [1780](https://github.com/OpenTracksApp/OpenTracks/issues/1780)

closed 2023-12-09 by dennisguse