OpenTracksApp/OpenTracks #1780: Bluetooth sensors cause crash

Mats391, 2023-12-09:
**Describe the bug**
When using Bluetooth sensors for heart rate or speed+distance, the app crashes. It happens when ending the recording, but sometimes also during the recording.
I tested 
* heart rate
* speed+distance
* cadence

Cadence sensor did not cause the crash, but was also not recording anything. The speed+distance did also not record anything. Neither of those sensors showed that they were connected in the recording screen. The heart rate did show and record correctly.

**To Reproduce**
1. Add Bluetooth sensor
2. Start recording
3. Wait a bit
4. End recording

If applicable:
Crash from heart rate:
[image]

Crash from speed+distance
[image]

**Technical information**
 - Device: Pixel 6A
 - OS: CalyxOs
 - OpenTracks version: 4.9.6-2
 - OpenTracks commit id (only for nightly builds): g2412a226

dennisguse, 2023-12-09:
Damn; that seems to be the same "problem" as #1779 and shouldn't happen.
Somehow data is still processed after the recording is stopped.

Geier0815, 2023-12-20:
At my side, using the v4.9.7 Version Code: 5559 with a Polar H10, the app ends by itself and stop recording after around 800 meter riding a bike. I guess that this is around the time when the screen goes dark. Permission for Location (Standort) is set to "while using".

Without the H10 the app works as expected.

Nokia 4.2 Android one 11

dennisguse, 2023-12-20:
@Geier0815 interesting finding. 
With "ends itself" you mean a crash?

Geier0815, 2023-12-20:
@dennisguse.

I can't say for sure that it was a crash. After the end of my riding I unlocked the screen and found my normal viewing as if no app was started. Opentrack was closed and there was a recording of the mentioned around 800 meter, if I started the app then.
If it were crashed I expected an error message or leftovers of the app or something similar, so I can't tell you for sure. This happened two times today, but the problem is, I'm owning the H10 since today morning, so it's not a longer observation.

Geier0815, 2023-12-21:
@dennisguse 

today I've got an crash report:

### App information
* ID: de.dennisguse.opentracks
* Version: 5559 v4.9.7

### Device information
* Brand: Nokia
* Device: PAN_sprout
* Model: Nokia 4.2
* Id: RKQ1.200928.002
* Product: Panther_00EEA

### Firmware
* SDK: 30
* Release: 11
* Incremental: 00EEA_3_24B

### Cause of error
```java
Exception in thread "main": java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.sensors.sensorData.AggregatorHeartRate.add(de.dennisguse.opentracks.sensors.sensorData.Raw)' on a null object reference
at de.dennisguse.opentracks.sensors.sensorData.SensorDataSet.update(SensorDataSet.java:100)
at de.dennisguse.opentracks.sensors.SensorManager$1.onChange(SensorManager.java:38)
at de.dennisguse.opentracks.sensors.BluetoothHandlerManagerHeartRate.handlePayload(BluetoothHandlerManagerHeartRate.java:48)
at de.dennisguse.opentracks.sensors.BluetoothConnectionManager$1.onCharacteristicChanged(BluetoothConnectionManager.java:123)
at android.bluetooth.BluetoothGatt$1$8.run(BluetoothGatt.java:478)
at android.os.Handler.handleCallback(Handler.java:938)
at android.os.Handler.dispatchMessage(Handler.java:99)
at android.os.Looper.loop(Looper.java:223)
at android.app.ActivityThread.main(ActivityThread.java:7697)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:952)
```

Geier0815, 2023-12-21:
Next test:
With "keep display always on while recording" (Bildschirm eingeschaltet lassen) in the settings of opentracks, the app crashed when I stopped the recording. Error message like the one I posted already. BUT, it seems that the recording itself is ok.
The second test I've tried was to record with the HR sensor of my MI5 Band, which worked in former times. But there I get no HR. I've then started gadgetbridge and there the HR were shown and recorded. No idea what's going wrong there, I will test tomorrow again.

Mats391, 2023-12-25:
I tried it with the nightly containing the above change and this is the crash report when starting with bluetooth heart rate sensor. If needed, I can try it with the cadence sensor later. 
[image]

dennisguse, 2023-12-25:
@Mats391 I know - the crash is now always (but enforced).

See: https://github.com/OpenTracksApp/OpenTracks/issues/1792#issuecomment-1868927388
```
This error happens with the second start of a recording.
After the first recording, you have to "force close" OpenTracks via the systems manager.
```

dennisguse, 2023-12-25:
If anybody wants to test this: see #1793.
I will give this some more review time tomorrow and would create a new release if nobody stops me :smirk:

commit 6cd627e8 (2023-12-26): Bugfix: TrackRecordingService can only start sensors once.

Fixes #1780.
closed 2023-12-26 by dennisguse