OpenTracksApp/OpenTracks #1781: Recording fails in version 4.9.6

vlmendz, 2023-12-09:
**Please describe the problem:**

Only a few positions are recorded at the beginning and end of the recording (3-6 in total).

**To reproduce it:**

Try to record a track

...

**Checklist**

- [ ] Consulted [https://dontkillmyapp.com/]
- [ x ] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: [Samsung Galaxy A52s 5G]
- OS: [One UI 6.0, Android 14]
- OpenTracks version: [v4.9.6]
- OpenTracks commit id (only for nightly builds): [V4.9.6-1-GF80AAFOOE]

P.S. I suspect the problem is related to the One UI 6.0/Android 14 upgrade.
Previously I was using OT v4.8 3 in One UI 5.1 with no problems. I had to upgrade to v4.9.6 because 4.8.3 crashed when starting recording in One UI 6.0.

dennisguse, 2023-12-09:
Can you check the GPS / location permissions?

vlmendz, 2023-12-09:
Only nearby devices permission not granted

lbrdt, 2023-12-10:
I've been having this problem as well. On last 2 recordings, it only recorded less than 100m distance (around 84m and 62m respectively) and stopped, however I think it recorded the altitude gain and loss.

[image]

Additional info:
- Android 14
- Pixel 6a

I recently updated to December Security update on 7th. I don't know if it had anything to do with it. OpenTracks has been in "Optimized" battery usage mode but I didn't have this problem before.

dennisguse, 2023-12-10:
Following options:
1. OpenTracks v4.9.6 is broken
2. Android update/ upgrade broke something

As I have no Android14 device available, testing must be done you. :rocket: 

What to do:
1. Check that all permissions are granted (incl. location) [likely correct]
2. Check if toggle `battery optimized` is changing the behavior
3. Check if screen on or screen off has an impact
4. Downgrade to anything earlier than v.4.9.6

How to do:
1. Go outdoors
2. Start a recording
3. ... wait for GPS fix (e.g., coordinates or absolute altitude is shown)
4. Start walking 
5. ... distance should be updated (usually every 10m; depends on your settings)
6. Stop recording

For downgrading: Here you can find v4.9.0 PlayStore release: [5428.zip](https://github.com/OpenTracksApp/OpenTracks/files/13627217/5428.zip)
on). This can be installed in parallel to the F-Droid release.
If you are using the PlayStore version, please first export everything from the existing installation as may need to uninstall it first.

lbrdt, 2023-12-11:
I'll be going day after tomorrow. I'll report back

vlmendz, 2023-12-11:
I did some short tests with v4 9.6 with the default (optimized) battery setting and found the following:

**-With screen off:** 

 -All permissions granted: all ok. 

 -Permission nearby devices not granted: recording fails. 

**-With screen on:**

 -All permissions granted: all ok. 

 -Nearby devices permission not granted: all ok

Summarizing, for the recording to work ok with the screen off and the combination of OT v4.9.6 and One UI 6.0/Android 14, it is necessary to grant the Nearby devices permission.

dennisguse, 2023-12-12:
This is then related to #1749.
At least it doesn't crash and it is not a power management situation ;)

Problem seems to be here: https://github.com/OpenTracksApp/OpenTracks/blob/24212a226ec07765955b9cfaf925382daad2e413/src/main/java/de/dennisguse/opentracks/services/TrackRecordingService.java#L206

If the sensors cannot be started, the TrackRecordingService doesn't start at all.
Best guess: since the notification is now optional (i.e., no permission, no notification), this is not visible to the user.

Two things: 
1. We should do this permission check earlier (even crashing is silently hiding the problem)
2. The UI needs to react to that properly

Bad news: I am really busy at the moment, so I cannot fix this soonish.

lbrdt, 2023-12-14:
Here's my last run with "unrestricted" battery usage mode. It had the same problem and stopped recording 1 minutes in. I could see OpenTracks on the notification bar but going in the app, the record button is reset and not showing a "stop" sign anymore.

https://i.imgur.com/ctYdsB3.png

OpenTracks does have access to "Nearby Devices" permission for me and still didn't work as intended.

https://i.imgur.com/OmDRPMq.png

dennisguse, 2023-12-16:
@lbrdt what you are now describing sounds like some kind of crash.
Could get a stacktrace / logcat?

lbrdt, 2023-12-31:
> @lbrdt what you are now describing sounds like some kind of crash. Could get a stacktrace / logcat?

Sorry for the late reply. Been busy with the holidays. Here's what I got from the logcat. I hope it helps

`12-09 06:03:07.060 E/AndroidRuntime(24117): FATAL EXCEPTION: main
12-09 06:03:07.060 E/AndroidRuntime(24117): Process: de.dennisguse.opentracks, PID: 24117
12-09 06:03:07.060 E/AndroidRuntime(24117): java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.sensors.sensorData.AggregatorHeartRate.add(de.dennisguse.opentracks.sensors.sensorData.Raw)' on a null object reference
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at de.dennisguse.opentracks.sensors.sensorData.SensorDataSet.update(SensorDataSet.java:100)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at de.dennisguse.opentracks.sensors.SensorManager$1.onChange(SensorManager.java:38)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at de.dennisguse.opentracks.sensors.BluetoothHandlerManagerHeartRate.handlePayload(BluetoothHandlerManagerHeartRate.java:48)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at de.dennisguse.opentracks.sensors.BluetoothConnectionManager$1.onCharacteristicChanged(BluetoothConnectionManager.java:123)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at android.bluetooth.BluetoothGattCallback.onCharacteristicChanged(BluetoothGattCallback.java:159)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at android.bluetooth.BluetoothGatt$1$8.run(BluetoothGatt.java:578)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at android.os.Handler.handleCallback(Handler.java:958)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at android.os.Handler.dispatchMessage(Handler.java:99)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at android.os.Looper.loopOnce(Looper.java:205)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at android.os.Looper.loop(Looper.java:294)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at android.app.ActivityThread.main(ActivityThread.java:8194)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at java.lang.reflect.Method.invoke(Native Method)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
12-09 06:03:07.060 E/AndroidRuntime(24117): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
12-09 07:31:03.538 E/AndroidRuntime(24339): FATAL EXCEPTION: main
12-09 07:31:03.538 E/AndroidRuntime(24339): Process: de.dennisguse.opentracks, PID: 24339
12-09 07:31:03.538 E/AndroidRuntime(24339): java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.sensors.sensorData.AggregatorBarometer.add(de.dennisguse.opentracks.sensors.sensorData.Raw)' on a null object reference
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at de.dennisguse.opentracks.sensors.sensorData.SensorDataSet.update(SensorDataSet.java:123)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at de.dennisguse.opentracks.sensors.SensorManager$1.onChange(SensorManager.java:38)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at de.dennisguse.opentracks.sensors.GainManager.onSensorValueChanged(GainManager.java:44)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at de.dennisguse.opentracks.sensors.driver.BarometerInternal.onSensorChanged(BarometerInternal.java:31)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at android.hardware.SystemSensorManager$SensorEventQueue.dispatchSensorEvent(SystemSensorManager.java:1011)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at android.os.MessageQueue.nativePollOnce(Native Method)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at android.os.MessageQueue.next(MessageQueue.java:335)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at android.os.Looper.loopOnce(Looper.java:162)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at android.os.Looper.loop(Looper.java:294)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at android.app.ActivityThread.main(ActivityThread.java:8194)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at java.lang.reflect.Method.invoke(Native Method)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:552)
12-09 07:31:03.538 E/AndroidRuntime(24339): 	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:971)
12-10 06:02:53.060 E/AndroidRuntime(31299): FATAL EXCEPT
ION: main
12-10 06:02:53.060 E/AndroidRuntime(31299): Process: de.dennisguse.opentracks, PID: 31299
12-10 06:02:53.060 E/AndroidRuntime(31299): java.lang.NullPointerException: Attempt to invoke virtual method 'void de.dennisguse.opentracks.sensors.sensorData.AggregatorHeartRate.add(de.dennisguse.opentracks.sensors.sensorData.Raw)' on a null object reference
12-10 06:02:53.060 E/AndroidRuntime(31299): 	at de.dennisguse.opentracks.sensors.sensorData.SensorDataSet.update(SensorDataSet.java:100)
12-10 06:02:53.060 E/AndroidRuntime(31299): 	at de.dennisguse.opentracks.sensors.SensorManager$1.onChange(SensorManager.java:38)
12-10 06:02:53.060 E/AndroidRuntime(31299): 	at de.dennisguse.opentracks.sensors.BluetoothHandlerManagerHeartRate.handlePayload(BluetoothHandlerManagerHeartRate.java:48)
12-10 06:02:53.060 E/AndroidRuntime(31299): 	at de.dennisguse.opentracks.sensors.BluetoothConnectionManager$1.onCharacteristicChanged(BluetoothConnectionManager.java:123)
12-10 06:02:53.060 E/AndroidRuntime(31299): 	at android.bluetooth.BluetoothGattCallback.onCharacteristicChanged(BluetoothGattCallback.java:159)
[... cut ...]
12-31 06:05:46.176 I/trusty  (    0): [GF_TA][E][gf_algo_utils][gf_is_gain_decreased_retry0] gain ratio: 100
12-31 06:05:46.176 I/trusty  (    0): [GF_TA][I][gf_cali][cali_process] authenticate, sNum = 1
12-31 06:05:46.176 I/trusty  (    0): [GF_TA][I][gf_cali][cali_process] stRawdata.psData=0x***, stRawdata.psDarkImage=0, stRawdata.sNum=1
12-31 06:05:46.176 I/trusty  (    0): [GF_TA][I][gf_cali][cali_process] nSegThre=-1
12-31 06:05:46.176 I/trusty  (    0): [GF_TA][I][gf_cali][cali_process] nCaliMode=2, support_moire_filter=1
12-31 06:05:46.176 I/trusty  (    0): [GF_TA][I`

lbrdt, 2023-12-31:
Here's one that I ran after cleaning the logcat and then just running OpenTracks.
```12-31 10:32:18.064 D/StatsLog(27535): LAUNCHER_TASK_LAUNCH_TAP
12-31 10:32:18.064 V/WindowManagerShell(25185): Transition requested: android.os.BinderProxy@4db4100 TransitionRequestInfo { type = 3, triggerTask = TaskInfo{userId=0 taskId=63929 displayId=0 isRunning=true baseIntent=Intent { flg=0x1400c000 cmp=de.dennisguse.opentracks/.TrackListActivity } baseActivity=ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity} topActivity=ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity} origActivity=null realActivity=ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity} numActivities=1 lastActiveTime=2059263453 supportsMultiWindow=true resizeMode=1 isResizeable=true minWidth=-1 minHeight=-1 defaultMinSize=220 token=WCT{android.window.IWindowContainerToken$Stub$Proxy@528dd39} topActivityType=1 pictureInPictureParams=null shouldDockBigOverlays=false launchIntoPipHostTaskId=-1 lastParentTaskIdBeforePip=-1 displayCutoutSafeInsets=Rect(0, 132 - 0, 0) topActivityInfo=ActivityInfo{8e3bf7e de.dennisguse.opentracks.TrackListActivity} launchCookies=[] positionInParent=Point(0, 0) parentTaskId=-1 isFocused=true isVisible=false isVisibleRequested=false isSleeping=false topActivityInSizeCompat=false topActivityEligibleForLetterboxEducation= false isLetterboxDoubleTapEnabled= false topActivityEligibleForUserAspectRatioButton= false topActivityBoundsLetterboxed= false isFromLetterboxDoubleTap= false topActivityLetterboxVerticalPosition= -1 topActivityLetterboxHorizontalPosition= -1 topActivityLetterboxWidth=-1 topActivityLetterboxHeight=-1 isUserFullscreenOverrideEnabled=false isTopActivityTransparent=false locusId=null displayAreaFeatureId=1 cameraCompatControlState=hidden}, remoteTransition = RemoteTransition { remoteTransition = android.window.IRemoteTransition$Stub$Proxy@6c51cdf, appThread = android.app.IApplicationThread$Stub$Proxy@4336d2c, debugName = LaunchFromRecents }, displayChange = null, flags = 0 }
12-31 10:32:18.064 V/WindowManagerShell(25185): RemoteTransition directly requested for android.os.BinderProxy@4db4100: RemoteTransition { remoteTransition = android.window.IRemoteTransition$Stub$Proxy@6c51cdf, appThread = android.app.IApplicationThread$Stub$Proxy@4336d2c, debugName = LaunchFromRecents }
12-31 10:32:18.067 D/WindowManagerShell(25185): onActivityRestartAttempt: ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity}
12-31 10:32:18.069 W/ProcessStats( 1641): Tracking association SourceState{ece06c2 kl.ime.oh/10285 ImpBg #1853193} whose proc state 6 is better than process ProcessState{da495d7 klye.usertext/10287 pkg=klye.usertext} proc state 15 (1328 skipped)
12-31 10:32:18.072 V/SnapshotDrawerUtils(25185): Drawing snapshot surface sizeMismatch=true

12-31 10:32:18.083 V/WindowManager( 1641): Sent Transition #15509 createdAt=12-31 10:32:18.061 via request=TransitionRequestInfo { type = 3, triggerTask = TaskInfo{userId=0 taskId=63929 displayId=0 isRunning=true baseIntent=Intent { flg=0x1400c000 cmp=de.dennisguse.opentracks/.TrackListActivity } baseActivity=ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity} topActivity=ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity} origActivity=null realActivity=ComponentInfo{de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity} numActivities=1 lastActiveTime=2059263453 supportsMultiWindow=true resizeMode=1 isResizeable=true minWidth=-1 minHeight=-1 defaultMinSize=220 token=WCT{RemoteToken{f4c3525 Task{92f6397 #63929 type=standard A=10242:de.dennisguse.opentracks}}} topActivityType=1 pictureInPictureParams=null shouldDockBigOverlays=false launchIntoPipHostTaskId=-1 lastParentTaskIdBeforePip=-1 displayCutoutSafeInsets=Rect(0, 132 - 0, 0) topActivityInfo=ActivityInfo{b4216fa de.dennisguse.opentracks.TrackListActivity} launchCookies=[] positionInParent=Point(0, 0) parentTaskId=-1 isFocused=true isVisible=false isVisibleRequested=false isSleeping=false topActivityInSizeCompat=false topActivityEligibleForLetterboxEducation= false isLetterboxDoubleTapEnabled= false topActivityEligibleForUserAspectRatioButton= false topActivityBoundsLetterboxed= false isFromLetterboxDoubleTap= false topActivityLetterboxVerticalPosition= -1 topActivityLetterboxHorizontalPosition= -1 topActivityLetterboxWidth=-1 topActivityLetterboxHeight=-1 isUserFullscreenOverrideEnabled=false isTopActivityTransparent=false locusId=null displayAreaFeatureId=1 cameraCompatControlState=hidden}, remoteTransition = RemoteTransition { remoteTransition = android.window.IRemoteTransition$Stub$Proxy@c2b90e8, appThread = android.app.IApplicationThread$Stub$Proxy@239301, debugName = LaunchFromRecents }, displayChange = null, flags = 0 }
12-31 10:32:18.083 V/WindowManager( 1641):     startWCT=WindowContainerTransaction { changes = {} hops = [] errorCallbackToken=null taskFragmentOrganizer=null }
12-31 10:32:18.083 V/WindowManager( 1641):     info={id=15509 t=TO_FRONT f=0x0 trk=0 r=[0@Point(0, 0)] c=[{WCT{RemoteToken{f4c3525 Task{92f6397 #63929 type=standard A=10242:de.dennisguse.opentracks}}} m=TO_FRONT f=MOVE_TO_TOP leash=Surface(name=Task=63929)/@0x6552e8f sb=Rect(0, 0 - 1080, 2400) eb=Rect(0, 0 - 1080, 2400) d=0},{WCT{RemoteToken{6dc6fb Task{c864ca6 #63903 type=standard A=10357:com.tananaev.logcat}}} m=TO_BACK f=NONE leash=Surface(name [... cut ...]
12-31 10:32:31.236 D/AOC     ( 1415): A3:MSG: usf_vsc_log.cc, 44: USF: [single-tap-prox-gated] configuring sensor with interval: 1, latency: 0
12-31 10:32:31.236 D/AOC     ( 1415): A3:MSG: vsc_controller.cc, 532: VSC: [Proximity Gated Single Tap Gesture] requestSensorEvents: sensor 0x10008, period=1
12-31 10:32:31.236 D/nodomain.freeyourgadget.gadgetbridge.externalevents.NotificationListener(21072): No Notification Filter found
12-31 10:32:31.237 D/AOC     ( 1415): A3:MSG: usf_sensor.cc, 1545: USF: Received start sampling request for```

closed 2024-04-01 by dennisguse

dennisguse, 2024-04-13:
@serrq if the permissions are not allowed, OpenTracks will not record. It will just show you the message and that's it. 
It is not the nicest looking solution, but should be sufficient for now.

commit 4bfd65bb (2024-04-01): Bugfix: handle missing permissions properly.

Affected only SDK >= 33.

Fixes #1781.