OpenTracksApp/OpenTracks #1787: Cadence sensor

janzen42, 2023-12-14:
**Describe the bug**
I have a Coospo BK467. This sensor can be used as a speed sensor or cadence sensor. I have set it to cadence mode and I have added it as a Cadence sensor in the app settings however when I record an activity, the Cadence data field shows a dash for the device hardware address and a dash for the rpm. I have tested the sensor and it works as expected using the CoospoRide app. 

**To Reproduce**
1. Go to Settings, Bluetooth Sensors, Cadence, select BK467
2. Add Cadence data field to layout and record activity.
3. Rotate the cadence sensor on bike.
4. Observe no readings in the cadence data field.

**Screenshots**
[image]
[image]

**Technical information**
 - Device: [e.g. Moto G7 Power]
 - OS: [e.g. LineageOS 20-20231013-mircoG-ocean]
 - OpenTracks version: [e.g. v4.9.6]

dennisguse, 2023-12-15:
The UI feedback is that OpenTracks did not connect to the sensor or that the sensor.is not configured (hence no address shown below the value). 
You may verify that the sensor is working with speed?

janzen42, 2023-12-15:
Yes, I just set it up as a speed sensor and it worked for speed. (I am testing using a stationary bike so I assume the speed isn't due to changes in gps location.)

dennisguse, 2023-12-15:
How did you set the sensor into cadence mode beside attaching it to the crank?
The BLE standard for these kind sensors allows to send two values (one for crank and one for wheel) that are reported.
I guess back then the assumption was that one sensor may be able to measure both.
So, if the sensor is reporting it's measurement as wheel rotation; OpenTracks will only handle it as wheel rotation.

Code for cycling cadence is in BluetoothHandlerCyclingCadence.

janzen42, 2023-12-15:
I do wonder what the sensor reports. I'm not sure how to dig into that. The sensor has two distinct modes that can be set by removing and replacing the battery. The sensor flashes a red light to indicate it is in speed mode and a blue light to indicate cadence mode. I believe the two modes are critical to it functioning correctly since when in speed mode it is mounted to the hub of the wheel and in cadence mode it is mounted to the crankshaft arm.

dennisguse, 2023-12-15:
I see. 
Can you compile+start Open tracks via AndroidStudio?
If so, you can log the sensor output and check what it is doing.

dennisguse, 2023-12-16:
And you may check if the sensor in cadence mode is working as a speed sensor for OpenTracks.

korn806, 2023-12-16:
I can confirm the existance of this bug since the v4.9.6 update. My cadence sensor worked before that and is still recognized and works in other Apps without a problem.

dennisguse, 2023-12-17:
@korn806 thanks. 
Then I broke something....

dennisguse, 2023-12-17:
Can you check if #1789 fixes the problem?

janzen42, 2023-12-17:
I am trying but I'm quite unfamiliar with this process. I'm not sure where to start. Is there a tutorial you can recommend?

janzen42, 2023-12-17:
I managed to download the project and loaded it into Studio but now I'm getting a Gradle error so I'm working on figuring out why that is happening. 

> Gradle DSL method not found: 'setVersionCodeOverride()'
> 
> Possible causes:
> 
> Your project may be using a version of the Android Gradle plug-in that does not contain the method (e.g. 'testCompile' was added in 1.1.0).

janzen42, 2023-12-17:
I managed to download the project and loaded it into Studio but now I'm getting a Gradle error and I can't figure it out. There must be something setup wrong in Visual Studio but I don't know what it could be.

`Build file '/home/janzen42/Downloads/OpenTracks-cadence-1787/build.gradle' line: 119

A problem occurred configuring root project 'OpenTracks-cadence-1787'.
> org.codehaus.groovy.runtime.metaclass.MissingMethodExecutionFailed: No signature of method: com.android.build.gradle.internal.api.ApkVariantOutputImpl.setVersionCodeOverride() is applicable for argument types: (null) values: [null]
  Possible solutions: setVersionCodeOverride(int), getVersionCodeOverride(), setVersionNameOverride(java.lang.String), getVersionNameOverride()

* Try:
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

* Exception is:
org.gradle.api.ProjectConfigurationException: A problem occurred configuring root project 'OpenTracks-cadence-1787'.
	at org.gradle.configuration.project.LifecycleProjectEvaluator.wrapException(LifecycleProjectEvaluator.java:84)
	at org.gradle.configuration.project.LifecycleProjectEvaluator.addConfigurationFailure(LifecycleProjectEvaluator.java:77)
	at org.gradle.configuration.project.LifecycleProjectEvaluator.access$500(LifecycleProjectEvaluator.java:55)
	at org.gradle.configuration.project.LifecycleProjectEvaluator$NotifyAfterEvaluate.run(LifecycleProjectEvaluator.java:255)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:157)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.run(DefaultBuildOperationRunner.java:47)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:68)
	at org.gradle.configuration.project.LifecycleProjectEvaluator$EvaluateProject.lambda$run$0(LifecycleProjectEvaluator.java:114)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$applyToMutableState$1(DefaultProjectStateRegistry.java:406)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.lambda$fromMutableState$2(DefaultProjectStateRegistry.java:429)
	at org.gradle.internal.work.DefaultWorkerLeaseService.withReplacedLocks(DefaultWorkerLeaseService.java:360)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.fromMutableState(DefaultProjectStateRegistry.java:429)
	at org.gradle.api.internal.project.DefaultProjectStateRegistry$ProjectStateImpl.applyToMutableState(DefaultProjectStateRegistry.java:405)
	at org.gradle.configuration.project.LifecycleProjectEvaluator$EvaluateProject.run(LifecycleProjectEvaluator.java:100)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:29)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$1.execute(DefaultBuildOperationRunner.java:26)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:66)
	at org.gradle.internal.operations.DefaultBuildOperationRunner$2.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:157)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.execute(DefaultBuildOperationRunner.java:59)
	at org.gradle.internal.operations.DefaultBuildOperationRunner.run(DefaultBuildOperationRunner.java:47)
	at org.gradle.internal.operations.DefaultBuildOperationExecutor.run(DefaultBuildOperationExecutor.java:68)
	at org.gradle.configuration.project.LifecycleProjectEvaluator.evaluate(LifecycleProjectEvaluator.java:72)
	at org.gradle.api.internal.project.DefaultProject.evaluate(DefaultProject.java:790)
	at org.gradle.api.internal.project.DefaultProject.evaluate(DefaultProject.java:156)
	at org.gradle.api.internal.project.ProjectLifecycleController.lambda$ensureSelfConfigured$2(ProjectLifecycleController.java:84)
	at org.gradle.internal.model.StateTransitionController.lambda$doTransition$14(StateTransitionController.java:255)
	at org.gradle.internal.model.StateTransitionController.doTransition(StateTransitionController.java:266)
	at org.gradle.internal.model.StateTransitionController.doTransition(StateTransitionController.java:254)
	at org.gradle.internal.model.StateTransitionController.lambda$maybeTransitionIfNotCurrentlyTransitioning$10(StateTransitionController.java:199)
	at org.gradle.internal.work.DefaultSynchronizer.withLock(DefaultSynchronizer.java:34)
	at org.gradle.internal.model.StateTransitionController.maybeTransitionIfNotCurrentlyTransitioning(StateTransitionController.java:195)
	at org.gradle.api.internal.project.ProjectLifecycleController.ensureSelfConfigured(ProjectLifecycleController.java:84)
[... cut ...]
	at org.gradle.launcher.daemon.server.DaemonStateCoordinator$1.run(DaemonStateCoordinator.java:297)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
	at org.gradle.internal.concurrent.AbstractManagedExecutor$1.run(AbstractManagedExecutor.java:47)
`
The second eorr is:
> Gradle DSL method not found: 'setVersionCodeOverride()'
> 
> Possible causes:
> 
> Your project may be using a version of the Android Gradle plug-in that does not contain the method (e.g. 'testCompile' was added in 1.1.0).

dennisguse, 2023-12-17:
@janzen42 quite good :D
Code works for me... - can you try to remove line 114:124?
It is just for building the nightly builds.
So, removing this part and check what happens:
```
    applicationVariants.configureEach { variant ->
        variant.resValue "string", "applicationId", variant.applicationId

        if (variant.buildType.name == 'nightly') {
            variant.outputs.configureEach {
                setVersionCodeOverride(getVersionCode())
                setVersionNameOverride(getVersionName())
                outputFileName = "${applicationId}_${variant.versionCode}.apk"
            }
        }
    }
```

janzen42, 2023-12-17:
I got a successful build. What should I do next? 

> Starting Gradle Daemon...
> Gradle Daemon started in 503 ms
> 
> > Configure project :
> fatal: not a git repository (or any parent up to mount point /)
> Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
> 
> > Task :prepareKotlinBuildScriptModel UP-TO-DATE
> 
> Deprecated Gradle features were used in this build, making it incompatible with Gradle 9.0.
> 
> You can use '--warning-mode all' to show the individual deprecation warnings and determine if they come from your own scripts or plugins.
> 
> For more on this, please refer to https://docs.gradle.org/8.5/userguide/command_line_interface.html#sec:command_line_warnings in the Gradle documentation.
> 
> BUILD SUCCESSFUL in 7s
>

janzen42, 2023-12-17:
I tried to Run and I got this error.

`Executing tasks: [:assembleDebug] in project /home/janzen42/Downloads/OpenTracks-cadence-1787

> Configure project :
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

> Task :createDebugVariantModel
> Task :preBuild UP-TO-DATE
> Task :preDebugBuild UP-TO-DATE
> Task :mergeDebugNativeDebugMetadata NO-SOURCE
> Task :generateDebugResValues
> Task :dataBindingMergeDependencyArtifactsDebug
> Task :extractDebugSupportedLocales
> Task :generateDebugResources
> Task :generateDebugLocaleConfig
> Task :generateDebugBuildConfig
> Task :mergeDebugResources
> Task :packageDebugResources
> Task :javaPreCompileDebug
> Task :mapDebugSourceSetPaths
> Task :checkDebugAarMetadata
> Task :createDebugCompatibleScreenManifests
> Task :extractDeepLinksDebug
> Task :parseDebugLocalResources
> Task :processDebugMainManifest
> Task :dataBindingGenBaseClassesDebug
> Task :processDebugManifest
> Task :mergeDebugShaders
> Task :compileDebugShaders NO-SOURCE
> Task :generateDebugAssets UP-TO-DATE
> Task :mergeDebugAssets
> Task :compressDebugAssets
> Task :desugarDebugFileDependencies
> Task :processDebugJavaRes NO-SOURCE
> Task :mergeDebugJniLibFolders
> Task :checkDebugDuplicateClasses
> Task :mergeDebugNativeLibs NO-SOURCE
> Task :mergeDebugJavaResource
> Task :stripDebugDebugSymbols NO-SOURCE
> Task :mergeLibDexDebug
> Task :validateSigningDebug
> Task :writeDebugAppMetadata
> Task :writeDebugSigningConfigVersions
> Task :processDebugManifestForPackage
> Task :l8DexDesugarLibDebug
> Task :processDebugResources FAILED
> Task :mergeExtDexDebug

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':processDebugResources'.
> A failure occurred while executing com.android.build.gradle.internal.res.LinkApplicationAndroidResourcesTask$TaskAction
   > Android resource linking failed
     de.dennisguse.opentracks-main-46:/xml/settings.xml:65: error: resource string/applicationId (aka de.dennisguse.opentracks.debug:string/applicationId) not found.
     de.dennisguse.opentracks-main-46:/xml/settings_import_export.xml:13: error: resource string/applicationId (aka de.dennisguse.opentracks.debug:string/applicationId) not found.
     de.dennisguse.opentracks-main-46:/xml/settings_import_export.xml:27: error: resource string/applicationId (aka de.dennisguse.opentracks.debug:string/applicationId) not found.
     de.dennisguse.opentracks-main-46:/xml/settings_import_export.xml:34: error: resource string/applicationId (aka de.dennisguse.opentracks.debug:string/applicationId) not found.
     de.dennisguse.opentracks-main-46:/xml/settings_import_export.xml:41: error: resource string/applicationId (aka de.dennisguse.opentracks.debug:string/applicationId) not found.
     error: failed linking file resources.

* Try:
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':processDebugResources'.
[... 235 lines cut ...]
		at org.gradle.internal.operations.DefaultBuildOperationRunner.call(DefaultBuildOperationRunner.java:53)
		at org.gradle.internal.operations.DefaultBuildOperationExecutor.call(DefaultBuildOperationExecutor.java:73)
		at org.gradle.workers.internal.AbstractWorker.executeWrappedInBuildOperation(AbstractWorker.java:41)
		at org.gradle.workers.internal.NoIsolationWorkerFactory$1.execute(NoIsolationWorkerFactory.java:59)
		at org.gradle.workers.internal.DefaultWorkerExecutor.lambda$submitWork$0(DefaultWorkerExecutor.java:170)
		at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
		at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.runExecution(DefaultConditionalExecutionQueue.java:187)
		at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.access$700(DefaultConditionalExecutionQueue.java:120)
		at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner$1.run(DefaultConditionalExecutionQueue.java:162)
		at org.gradle.internal.Factories$1.create(Factories.java:31)
		at org.gradle.internal.work.DefaultWorkerLeaseService.withLocks(DefaultWorkerLeaseService.java:264)
		at org.gradle.internal.work.DefaultWorkerLeaseService.runAsWorkerThread(DefaultWorkerLeaseService.java:128)
		at org.gradle.internal.work.DefaultWorkerLeaseService.runAsWorkerThread(DefaultWorkerLeaseService.java:133)
		at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.runBatch(DefaultConditionalExecutionQueue.java:157)
		at org.gradle.internal.work.DefaultConditionalExecutionQueue$ExecutionRunner.run(DefaultConditionalExecutionQueue.java:126)
		at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:539)
		at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
		at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:64)
		at org.gradle.internal.concurrent.AbstractManagedExecutor$1.run(AbstractManagedExecutor.java:47)
		at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
		at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
		at java.base/java.lang.Thread.run(Thread.java:833)

BUILD FAILED in 6s
32 actionable tasks: 32 executed`

janzen42, 2023-12-17:
I can confirm this fixes the bug! I had downloaded the zip so I figured that was the reason for all the errors. I cloned the project using command 
`gh repo clone OpenTracksApp/OpenTracks -- --branch cadence#1787` and I made sure your change to line 72 of [BluetoothRemoteSensorManager.java](https://github.com/OpenTracksApp/OpenTracks/pull/1789/commits/64422e2f88f844612dbccd8ae5f2e62f94645604#diff-e938cb209f12c091f1471e66241445f49bf6d2bcc10eaf3
ef44a73cf51b474f8) was actually there since I wasn't sure if I had the right branch. That looked good so I compiled and I ran it by pairing over wifi to my Moto G7. As you can see in the screenshot attached we have success!
[image]

janzen42, 2023-12-17:
This is probably a separate issue but I noticed that the stats display shows the the rpm number continuously when cadence is over 60 rpm but when the cadence is lower it shows the number for about a second and then shows the dash. The lower the cadence, the more the dash is displayed rather than the last measured value.

dennisguse, 2023-12-17:
@janzen42 thanks for taking the time to test this.
I will craft a new release now :dagger: 

Regarding the <60 rpm: can you open a new issue for this?

commit 64422e2f (2023-12-17): Bugfix: cycling cadence data should be handled by BluetoothHandlerCyclingCadence.

Fixes #1787.

Introduced in 15f3ccbcbd0e2792a8e4540bca17210ae782ad3e
PR #1789 (2023-12-17): Bugfix: cycling cadence data should be handled by BluetoothHandlerCyc…
…lingCadence.

Fixes #1787.

Introduced in 15f3ccbcbd0e2792a8e4540bca17210ae782ad3e
commit d08552a0 (2023-12-17): Bugfix: cycling cadence data should be handled by BluetoothHandlerCyclingCadence.

Fixes #1787.

Introduced in 15f3ccbcbd0e2792a8e4540bca17210ae782ad3e
closed 2023-12-17 by dennisguse