OpenTracksApp/OpenTracks #1790: Cadence data persistence in the stats screen

janzen42, 2023-12-17:
**Describe the bug**
The cadence data displayed in the stats screen alternates between showing a dash and showing the rpm number when the rpm is less than 60 rpm.

**To Reproduce**
1. In Settings, Bluetooth Sensors, add a Cadence sensor
2. In Settings, User Interface, Customize your recording Layout, Add the Cadence data field.
3. Start recording. Vary the rpm to observe how the number is displayed continuously when the rpm is higher than 60. When the rpm is lower the rpm number is displayed but is replaced with the dash prior to the next reading being displayed. The lower the rpm, the more often the dash is observed.

If applicable:

https://github.com/OpenTracksApp/OpenTracks/assets/79019480/3b1be498-2f2f-4e76-a37a-f284f351c928

**Technical information**
 - Device: Moto G7 Power
 - OS: [LineageOS 20-20231013-mircoG-ocean]
 - OpenTracks version: [v4.9.7-debug/v4.9.7-3-g64422e2f8]

janzen42, 2023-12-17:
The cadence sensor used for this test was the Coospo BK467.

dennisguse, 2023-12-17:
Can you add this sensor to the sensor readme?

Let me think about what might be wrong :)

dennisguse, 2023-12-29:
I did some mental debugging while walking home.
It could be that the sensor is sending the data repeated and then the value is removed from OpenTracks.
This line would be responsible for this: https://github.com/OpenTracksApp/OpenTracks/blob/bb23bac7ec17345cacb82b86ef03c2443a13f98e/src/main/java/de/dennisguse/opentracks/sensors/sensorData/AggregatorCyclingCadence.java#L28

@janzen42 Can you remove this line and check if this solves the issue?

janzen42, 2023-12-30:
> I did some mental debugging while walking home. It could be that the sensor is sending the data repeated and then the value is removed from OpenTracks. This line would be responsible for this: [...]

I tried it out and it solves the issue however it creates a new issue. When one stops pedaling the last recorded cadence value sticks indefinitely so the value never drops to zero. Cadence is a tricky one isn't it?

dennisguse, 2023-12-30:
:+1: 
Good that this works!

The sensor data has a TTL of 5s.
So, if OpenTracks doesn't  any further data, it should use getNoneValue().
... or there is another bug in AggregatorCyclingCadence...

janzen42, 2023-12-30:
Yeah, it doesn't seem to be working. I can pedal for a bit, get it to record 60rpm and then I stop for a minute. The By time screen shows cadence remaining at 60rpm to the end of the recording and this also causes an inaccurate avg cadence value on the Stats screen.

closed 2024-01-01 by dennisguse

OMTDesign, 2024-06-07:
> When one stops pedaling the last recorded cadence value sticks indefinitely so the value never drops to zero.

I also have this issue on the latest version 4.12.4 version code 5788. I have CYCPLUS S/C3 sensors, and Opentracks will hold the last received cadence and Speed value instead of dropping to zero. @dennisguse 
[image]
[image]

OMTDesign, 2024-06-07:
[image]

OMTDesign, 2024-06-07:
https://github.com/OpenTracksApp/OpenTracks/assets/39666543/39e41b57-7f61-4e61-81d8-33958e322505

dennisguse, 2024-06-07:
@OMTDesign Can you open a new issue for this?
I am not sure that this is related to this issue.

commit 0584f05d (2024-01-01): Bugfix: Cycling Cadence and Distance sensor allow for repeated data.

Fixes #1790.