OpenTracksApp/OpenTracks #1794: Sub-1-kilometer distances are announced as 10x the distance with TTS

Giraut, 2023-12-25:
When Statistics Annoucements is set to announce total distance at given time intervals, when the distance is announced and it's under 1 km, the TTS announces 10 times that distance. Or said another way, instead of saying "zero-point-something", it just says "something"

Example: if the distance is 0.8 km, the announcer will say "Total distance: 8 kilometers"

-----
- OpenTrack v4.9.7 (version code 5559)
- TTS: RHVoice 1.8.0 (com.github.olga_yakovleva.rhvoice.android) / Language: English (default voice: SLT)
- OS: CalyxOS

dennisguse, 2023-12-26:
Seemingly the number is set to .8km and the dot is ignored by the TTS.

dennisguse, 2023-12-26:
With the standard voice synthesizer this _works as expected_ (i.e., `... distance zero dot eight kilometers).
Can you forward this issue to RHVoice?

Relevant code for testing (using OpenTracks `VoiceAnnouncementUtils`):
```
    @Test
    public void getAnnouncement_metric_speed_small() throws InterruptedException {
        TrackStatistics stats = new TrackStatistics();
        stats.setTotalDistance(Distance.of(100));
        stats.setTotalTime(Duration.ofMinutes(1));
        stats.setMovingTime(Duration.ofMinutes(1));
        stats.setMaxSpeed(Speed.of(5));

        // when
        String announcement = VoiceAnnouncementUtils.createStatistics(context, stats, UnitSystem.METRIC, true, null, null).toString();
        var a = VoiceAnnouncementUtils.createStatistics(context, stats, UnitSystem.METRIC, true, null, null);
        TTSManager manager = new TTSManager(context);
        manager.start();
        Thread.sleep(1000);
        manager.announce(a);
        Thread.sleep(7000);

        // then
        assertEquals("Total distance .1 kilometers. 1 minute. Average moving speed 6.0 kilometers per hour.", announcement);
    }
```

closed 2023-12-26 by dennisguse