OpenTracksApp/OpenTracks #1796: Bluetooth device name is not shown (but address)

dennisguse, 2023-12-26:
When using any BLE sensor that is also connected, the device name should be shown.
Instead the sensor address is shown.

Relevant code: `SensorInterface.createEmptySensorData(String address) {}`.
This must be extended to also pass the sensor name.

Likely introduced in refactoring / released as v4.9.6

Geier0815, 2023-12-26:
For me both names of my "MI5 Band" and my "Polar H10" are shown as expected at the "choose your device" screen. At the overview settings screen at "Generisch" "Herzfrequenz" only the address is shown but that was so since I've got my first device back in May '23. That's not new.

### App information
v4.9.8-8g81a78820c
Version Code: 5572

### Device information
* Brand: Nokia
* Device: PAN_sprout
* Model: Nokia 4.2
* Id: RKQ1.200928.002
* Product: Panther_00EEA

### Firmware
* SDK: 30
* Release: 11

closed 2023-12-26 by dennisguse

dennisguse, 2023-12-26:
@Geier0815 then I broke that feature somewhen earlier....

commit 8f5388ed (2023-12-26): While recording: show BLE sensor name instead of address.

Fixes #1796.