OpenTracksApp/OpenTracks #1799: Crash from Distance+Speed sensor

Mats391, 2023-12-28:
**Describe the bug**
I paused the recording on stop and when I resumed it, the app crashed shorty after. After restarting the app, it crashed again when starting the recording, but on the third time it worked fine. I could not reproduce it since. The crash seems similar to #1780 but this time it happened while recording, not when ending the recording.

If applicable:
Crash trace
[image]

**Technical information**
 - Device: Pixel 6a
 - OS: CalyxOs 5.2.0
 - OpenTracks version: 4.9.9-16 nightly
 - OpenTracks commit id (only for nightly builds): gfa362de67

dennisguse, 2023-12-30:
Guess: sensor disconnected.
And then more data was received...

dennisguse, 2023-12-31:
I reproduced the issue; seems to be a race condition.

dennisguse, 2024-01-01:
It isn't a race condition:

How to reproduce:
1. Start a recording with a BLE sensor (used a heartrate sensor)
2. ... wait until sensor data is shown in OpenTracks
3. Turn off sensor
4. Turn on sensor

Step 4. triggers the crash.

Problem is the onDisconnect callback: https://github.com/OpenTracksApp/OpenTracks/blob/5de2012452dd2436df748dcbd30eb67bb91b4d75/src/main/java/de/dennisguse/opentracks/sensors/BluetoothConnectionManager.java#L182-L183

As this triggers the removal of the Aggregator in SensorDataSet.
And it is never setup again...

commit d906a59a (2024-01-05): Bugfix: a BLE sensor disconnect should only replace the aggregator.

Fixes #1799.
PR #1810 (2024-01-05): Bugfix: a BLE sensor disconnect should only replace the aggregator.
Fixes #1799.