OpenTracksApp/OpenTracks #1801: Average speed of time interval as voice output

timror, 2023-12-30:
**Is your feature request related to a problem? Please describe.**
When exercising I normally use a voice feedback on time interval. This gives me only the option to get the average speed for the whole exercise duration. This makes it difficult to assess how I have performed in the previous interval. 

**Describe the solution you'd like**
I would like the option (toggle) to get the average speed of the time interval, with respect to the duration of that interval

**Describe alternatives you've considered**
I now use the average speed of the whole recording exercise.

hercek, 2023-12-31:
It is already there but named badly in UI. Use Lap speed/pace instead of Average speed.

Lap speed/pace is actually "average speed of the last interval" despite the fact that it is named "Lap speed/pace".  If I understand the code correctly then there is no lap detection in it. The only way how laps can be taken into account if is if you set announcer interval to distance which is the same as the length of your lap. There should not be any mention of laps in the UI since there is no lap detection. On the other side, I would not appreciate if "interval" adjective is added to the UI since it gets too verbose.

My opinion is that the settings tab should support only two types of measurements:
1) Average/total quantities which would refer to the whole track.
2) Quantities without any qualifier (adjective added to it) in which case they would refer to the last interval: e.g. if it is "Speed" then it would mean average speed of the last interval.

timror, 2024-01-02:
Tnx for the explanation. I've tried the app with the following settings (sorry it's in Dutch)

[image]

Now I only hear the elapsed time, but no spoken feedback of the speed. Is it possible the voice output is broken somewhere?
Returning to the previous setting (avg /speed) does give me voice output, so that setting works as expected.

hercek, 2024-01-03:
Interesting. It works for me when I'm using distance interval instead fo time interval and only "lap" speed is reported. I'll try something more similar to your settings.

Do you have "Use device speaker" checked?
Do you use "distance and speed" Bluetooth sensor?
Does GPS properly record your track? I.e. whey you ask to show a track on a map, does it show correctly?

timror, 2024-01-03:
> Do you have "Use device speaker" checked?

It is unchecked. Normally I hear the feedback through my earpieces (aux port) 

> Do you use "distance and speed" Bluetooth sensor?

I don't use an external Bluetooth device. I only use my phone GPS. 

> Does GPS properly record your track? I.e. whey you ask to show a track on a map, does it show correctly?

Yes, works very well with KML export to OrganicMaps

> What version do you use?

v.4.9.9

hercek, 2024-01-04:
OK, I tested how this behaves today on version 4.10.0. It is very likely the behaviour is the same as in 4.9.9.

You can change this issue to a bug report: 
"announcer's interval based quantities are not reported when there is only time interval setup"

The likely actual reason is that these quantities are computed only based on the distance interval set in the announcer. If one has setup only time based interval then interval ("lap") based quantities are not computed at all and therefore not reported at all. If you would have setup both time and distance based interval then the interval ("lap") based quantities would be stalled during time based report; i.e. the would be reported as they were computed in the last distance interval.

If you are not allowed to do this change then create a new issue and link to this one.

Somebody may resolve this eventually. You can use a distance based interval in the meantime. Set "Time interval" to "Off" and set some distance in "Distance interval" in the Announcer settings.

hercek, 2024-01-04:
One way how to resolve this issue without changing java code is to keep "Lap" string in the "distance interval" based announcer quantities and only change "Distance interval" string in the settings to "Lap length". I quite like this option. It has my vote.

Though it would be handy to be able to set any lap distance (and not only the preselected ones) in this case.

dennisguse, 2024-01-04:
@timror @hercek pretty good discussion!
Kudos!

@hercek can you create a PR for changing the strings?
The idea sounds pretty good :D

hercek, 2024-01-04:
I can change "Distance interval" string in options to "Lap length" string. It will make things more clear in the options.

But later. I have currently unfinished version of PR for #1805 and do not want to mix many things together.

PR #1819 (2024-01-10): clarify that announcer's "distance interval" corresponds to "lap length"
 This fixes #1801 the most easy way. It does not implement announcer's
 interval statistics over time intervals. Instead, it only clarifies
 that "lap" quantities in the announcer are related to lap length.
 Therefore they will not work if there is no lap length defined.

# Thanks for your contribution.

**Link to the the issue**
#1801

**License agreement**
By opening this pull request, you are providing your contribution under the _Apache License 2.0_ (see [LICENSE.md](LICENSE.md)).
commit 4a6f6526 (2024-01-10): clarify that announcer's "distance interval" corresponds to "lap length"
 This fixes #1801 the most easy way. It does not implement announcer's
 interval statistics over time intervals. Instead, it only clarifies
 that "lap" quantities in the announcer are related to lap length.
 Therefore they will not work if there is no lap length defined.
closed 2024-01-10 by dennisguse