OpenTracksApp/OpenTracks #1802: Crash when starting a track while BLE power sensor is enabled

hercek, 2023-12-30:
**Describe the bug**
OpenTracks crashes a few seconds after a track recording is started. At least BLE power cycling sensor must be enabled before the recording is started. I can simulate the bug with Magene PES P505 Base Power Meter. But I think it is not related to this particular power meter. I think any BLE cycling power meter will do.

**To Reproduce**
1. clean & new OpenTracks installation
2. run it and get over the initial introductory screens
3. go to settings before starting any track
4. enable metric units
5. default activity: cycling
6. UI theme: day
7. show stats on lock screen : on
8. keep screen on: on
9. full screen: on
10. GPS accuracy: 20m
11. idle threshold: 5s
12. pair and enable cadence BLE sensor
13. pair and enable power BLE sensor
14. enable idle announcements
15. announcement distance interval: 500 m
16. leave only lap speed announcement enabled
17. **it is likely that only point 13 is really needed from the points 4 to 16 above**
18. start a track recording
19. a crash will happen in a few seconds (very likely just after the very first cycling power read out)

**Comments**
The crash can be simulated in versions V4.10.0, V4.9.9, V4.9.7. It works in version V4.9.5. I did not try versions V4.9.8 and V4.9.6. It is likely the error was introduced in version V4.9.6 when a lot of changes were done to BluetoothConnectionManagerCyclingPower.java, maybe in revision f2594371d388b700b85f189892a7aceb422ae53c.
It seems to be a standard problem with botched implementation of Java generics which postpones type checking to run time.

**Technical information**
See the attached logs for the crashes. They contain everything needed.
[OpenTracksCrashV4.9.7.txt](https://github.com/OpenTracksApp/OpenTracks/files/13798806/OpenTracksCrashV4.9.7.txt)
[OpenTracksCrashV4.10.0.txt](https://github.com/OpenTracksApp/OpenTracks/files/13798807/OpenTracksCrashV4.10.0.txt)

dennisguse, 2023-12-30:
This may solve the problem https://github.com/OpenTracksApp/OpenTracks/issues/1802

... but the stacktrace shows another problem.
However, SensorDataSet:160 should be compile time checked.

hercek, 2023-12-31:
OK, it seems to be working.
I cannot replicated the bug any more when using branch bugfix#1802.
Thanks.

dennisguse, 2023-12-31:
Thanks for testing!

@hercek can you add your sensor to the doucmentation?
https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md

closed 2023-12-31 by dennisguse

hercek, 2023-12-31:
Done: https://github.com/OpenTracksApp/OpenTracks/pull/1804

PR #1803 (2023-12-31): Bugfix: process cycling power sensor data correctly.
Fixes #1802.
commit 310ac8c0 (2023-12-31): Bugfix: process cycling power sensor data correctly.

Fixes #1802.