OpenTracksApp/OpenTracks #1806: report cycling power in the announcer

hercek, 2023-12-31:
**Is your feature request related to a problem? Please describe.**
Sometimes I'm slacking while training. Getting reminders about my last interval power output might help to motivate me to train hard enough.

**Describe the solution you'd like**
Add reporting of the last interval cycling power to the announcer. Control the reporting from options.

I have a very dirty test code which implements power announcing in my branch:
https://github.com/hercek/OpenTracks/tree/powerAnnounce
It is not controlled by options. It is not finished. It may be wrong. It is my very first Java code ever (I typically do C++ and Haskell). Anyway it seems to work just like I like it: short, without units, always on.

dennisguse, 2024-01-01:
Nice!
Just create a PR with the relevant changes and I can support you in getting this done :)

hercek, 2024-01-01:
It is not finished yet:
* power reporting cannot be switched off from settings
* the power string is not properly localized (just hard coded in english now)

Do you want me to create a pull request even when it is not finished yet? I.e. will you finish it instead of me?

I can finish it myself and create pull request when it is done but it will likely take a month or more if I'm unlucky. I posted this early so that if somebody is in a hurry then he can copy code from my branch (or may be only get inspired by it) and finish it himself.

dennisguse, 2024-01-01:
If you open a PR, it is easier to communicate.
I (or somebody else) may be able to give you some hints if you are on the right track :)

hercek, 2024-01-02:
I think this should be done after https://github.com/OpenTracksApp/OpenTracks/issues/1805 so that power units are not being added there just to be removed afterwards.

If a PR should be made for unfinished code then I guess it should go to a non-main branch. I guess you should create powerAnnounce branch so that I can create PR for it with the code I have so far.

dennisguse, 2024-05-12:
Fixed by #1844

closed 2024-05-12 by dennisguse