OpenTracksApp/OpenTracks #1812: Announcer's interval based quantities are not reported when there is only time interval setup

timror, 2024-01-04:
**Describe the bug**
As discussed in #1801.

**To Reproduce**
With the following settings in the app:
[image]

You only hear the elapsed time, but no spoken feedback of the speed of the interval. 

**Technical information**
 - OpenTracks version: v.4.9.9

dennisguse, 2024-01-04:
Seemingly, `lastInterval` is null for time triggered events.
Relevant code is found here: https://github.com/OpenTracksApp/OpenTracks/blob/d038f5c33aa6178b1bf139e05c4d8da30f0724cb/src/main/java/de/dennisguse/opentracks/services/announcement/VoiceAnnouncementManager.java#L158

dennisguse, 2024-12-29:
Fixed after v4.17.5 aka already in the nightly.
Will be released in the upcoming days.

closed 2024-12-29 by dennisguse