OpenTracksApp/OpenTracks #1823: optional chart fill

hercek, 2024-01-12:
**Is your feature request related to a problem? Please describe.**
Charts can be very hard to parse when there is more data series displayed (e.g. 3 and more). 

**Describe the solution you'd like**
The most significant reason for this is the chart fill (alpha blended lighter fill below a chart curve). It decreases legibility **a lot** when there are many data series in the chart. Add an option to the settings which will allow users to switch off the chart fill.

**Describe alternatives you've considered**
Remove the chart fill completely from the code. I prefer this more but the chart fill increases chart legibility **slightly** when there is only one or maybe two data series shown. This option would keep settings cleaner.

**Additional context**
I will likely provide a PR for the proposed solution or the alternative solution if one of them is approved.

dennisguse, 2024-01-12:
:+1: 
Can you share a screenshot how it looks like?

hercek, 2024-01-13:
Ok, here are some samples. The older version does not contain also the accepted PR which moves x-axis label lower right. So the comparison is not 100% fair. But I do not have only one commit older APK ready.

**Portrait mode with chart fill**
[image]

**Portrait mode without chart fill**
[image]

**Landscape mode with chart fill**
[image]

**Landscape mode without chart fill**
[image]

I think the problem is best visible in the landscape mode: try to follow the power curve in the interval from 4.2 km to 4.5 km (around the 3rd pin). It is more or less impossible in the version with the chart fill. Or I'm pretty much blind in my advanced age :)

Please specify whether you want version controlled by an option or just removing the fill all the time. Maybe it could be done with a 3-state checkbox - the 3-rd state would use fill for charts with only one or two data series.

hercek, 2024-01-13:
Yes. I'll make a PR for this version: hiding fill when there are at least 3 data series visible.

commit 7ebb629b (2024-01-13): fill the area below a chart curve only when there are at most 2 curves
 This makes charts more readable when there are many curves in it.
 Fixes #1823
closed 2024-01-13 by dennisguse