OpenTracksApp/OpenTracks #1824: Unexpected crash after unlock phone

duhow, 2024-01-12:
### App information
* ID: de.dennisguse.opentracks
* Version: 5602 v4.10.0

### Device information
* Brand: samsung
* Device: dm3q
* Model: SM-S918B
* Id: UP1A.231005.007
* Product: dm3qxeea

### Firmware
* SDK: 34
* Release: 14
* Incremental: S918BXXU3BWJM

### Cause of error
```java
Exception in thread "main": android.app.RemoteServiceException$ForegroundServiceDidNotStartInTimeException: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{842e215 u0 de.dennisguse.opentracks/.services.TrackRecordingService}
	at android.app.ActivityThread.generateForegroundServiceDidNotStartInTimeException(ActivityThread.java:2315)
	at android.app.ActivityThread.throwRemoteServiceException(ActivityThread.java:2286)
	at android.app.ActivityThread.-$$Nest$mthrowRemoteServiceException(Unknown Source:0)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2611)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loopOnce(Looper.java:230)
	at android.os.Looper.loop(Looper.java:319)
	at android.app.ActivityThread.main(ActivityThread.java:8893)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:608)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)
Caused by: android.app.StackTrace: Last startServiceCommon() call for this service was made here
	at android.app.ContextImpl.startServiceCommon(ContextImpl.java:2023)
	at android.app.ContextImpl.startForegroundService(ContextImpl.java:1967)
	at android.content.ContextWrapper.startForegroundService(ContextWrapper.java:847)
	at android.content.ContextWrapper.startForegroundService(ContextWrapper.java:847)
	at androidx.core.content.ContextCompat$Api26Impl.startForegroundService(ContextCompat.java:1189)
	at androidx.core.content.ContextCompat.startForegroundService(ContextCompat.java:752)
	at de.dennisguse.opentracks.services.TrackRecordingServiceConnection.execute(TrackRecordingServiceConnection.java:164)
	at de.dennisguse.opentracks.publicapi.AbstractAPIActivity.onCreate(AbstractAPIActivity.java:40)
	at android.app.Activity.performCreate(Activity.java:8944)
	at android.app.Activity.performCreate(Activity.java:8913)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1456)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:4146)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4322)
	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:103)
	at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:139)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:96)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2685)
	... 7 more
```

dennisguse, 2024-01-12:
Is this reproducible and if so: how?

duhow, 2024-01-12:
It's the first time I have seen a crash from the app.
Unlocked the phone and found it.
I suspect that Gadgetbridge may have triggered it?

dennisguse, 2024-01-12:
Was any data recorded aka did OpenTracks crash immediately after triggering record?
And you triggered it via Gadgetbridge.
This implies the screen was off when you started the recording?

duhow, 2024-01-13:
I did not open OpenTracks app for weeks.
Just did some exercise with my Mi Band, get my phone 10 minutes later, unlock it, and find this crash report.

dennisguse, 2024-01-13:
OK. Could you try if this is always crashing or only sometimes?
I don't have this setup at hand.

bentolor, 2024-02-13:
Thank you very much, Dennis, for your valueable tool. This helps me to increase the value of my otherwise rather dumb Mi Band 5. I've got the same/similar error: I see a full-screen error screen with an OpenTracks error. 

I assume this happens whenever I move and my paired Mi Band 5 autodetect walking as activity: This causes a request to Gadgetbridge to start a recording. Of course the phone is locked at this moment.

**This reproducible:** @dennisguse 
 1. Lock phone screen and initiate a GPS-based activity on the Mi Band (i.e. Cycling). 
 1. Stop it directly before actually starting the activity. 
 1. After phone screen unlock, crash screen is displayed.

### App information
* ID: de.dennisguse.opentracks
* Version: 5641 v4.11.0

### Device information
* Brand: OnePlus
* Device: OnePlus3T
* Model: ONEPLUS A3003
* Id: RQ3A.211001.001
* Product: lineage_oneplus3

### Firmware
* SDK: 30
* Release: 11
* Incremental: 417968da2c

### Cause of error
```java
Exception in thread "main": android.app.RemoteServiceException: Context.startForegroundService() did not then call Service.startForeground(): ServiceRecord{c7c0eb u0 de.dennisguse.opentracks/.services.TrackRecordingService}
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2005)
	at android.os.Handler.dispatchMessage(Handler.java:106)
	at android.os.Looper.loop(Looper.java:223)
	at android.app.ActivityThread.main(ActivityThread.java:7664)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
```

bentolor, 2024-02-14:
Thanks @dennisguse for your quick reaction! 

I just installed the nightly APK attached to the merge request, installed it on my phone, changed the package name in GB to `*.nightly` and then successfully executed the testcase above without any errors.

Buuuuut: Then I revert my changes above and tried to reproduce and… was not able to reproduce. 

Since our last conversation my F-Droid upgraded OT from v4.11.0 to v4.11.1: Could this patch already have fixed the issue? Otherwise it seems my reproduce is only stable under some circumstances and I'm unable to actually confirm or deny the fix is working.

Not sure how to proceed from here on: Is it safe to merge the change anyway?

dennisguse, 2024-02-14:
v4.11.1 is broken and v4.11.2 reverts the initial fix.
The recording was just stopped when the screen is turned as the recording service was started incorrectly. 

This PR contains a (hopefully) correct fix.
Can you try it for a real recording (outdoor, screen off, more than 10min; started and stopped via watch)?

bentolor, 2024-02-14:
Let me check I'm able to reproduce. Might take some time until I get the opportunity. Thank you for your support!

closed 2024-02-22 by dennisguse

bentolor, 2024-02-23:
Sorry for my lack of feedback @dennisguse , but I struggled to actually replicate the issue stably, even I was able to reproduce it stably when I wrote my comment above.

So it seems some additional state comes into play. I've successfully run nightly and yet not observed the issue, but I cannot confirm it really addresses the issue. So hope this works.

Thanks for your support and finger-crossed this solves the issue.

dennisguse, 2024-02-24:
@bentolor Thanks.
Hope it is solved; if it occurs again, just re-open the issue :D

commit 15d34f12 (2024-02-13): Bugfix: PublicAPI stop should not start TrackRecordingService in foreground.

Actually, TrackRecordingService will only be start in foreground if needed.

Fixes #1824.
commit 1bd3d663 (2024-02-13): Bugfix: PublicAPI stop should not start TrackRecordingService in foreground.

Actually, TrackRecordingService will only be start in foreground if needed.

Fixes #1824.
commit d0874f5e (2024-02-21): Bugfix: PublicAPI stop should not start TrackRecordingService in foreground.

Actually, TrackRecordingService will only be start in foreground if needed.

Fixes #1824.
commit e02dbd8c (2024-02-21): Bugfix: PublicAPI stop should not start TrackRecordingService in foreground.

Actually, TrackRecordingService will only be start in foreground if needed.

Fixes #1824.