OpenTracksApp/OpenTracks #1827: markers imported from a KMZ file which are at the end of the track are shown at the track start

hercek, 2024-01-13:
**Describe the bug**
Import of a KMZ file with markers which are located **after the start** of a track will lead to track charts with the markers only shown **at the start** of the track.

It is interesting that exporting this imported track again leads to a KMZ file which is the same as the imported KMZ file. So the error is somewhere in the OpenTrack internals and does not propagate again to the exports. Searching for the affected markers in the both the imported and again exported KMZ file shows that the markers point to the right places (somewhere far from the start of the track).

Markers are properly shown for tracks recorded directly in the app. This is related only to the imported tracks. Looks like possibly related to comparing floating point numbers precisely (while they differ slightly due to import/export) or maybe some timezone/localTime/GMT related stuff.

I do not care about this bug nor whether it ever fixed. I only wanted it recorded. I detected it by a chance while implementing #1820. That task was marginally related to marker drawing in the charts.

**To Reproduce**
1. Record a track with markers which are not at the very beginning of the track.
2. Export as KMZ.
3. Delete the track.
4. Import the KMZ.
5. Look at a chart of the imported track.
6. All markers will be shown (one over another) at the beginning of the track.

**Technical information**
 - This is likely unrelated to a specific version of Android.  I have seen it in the Android Pixel virtual phone of the current android-studio.
 -  The KMZ file was from my Samsung XCover4s with Android 11.
 - OpenTracks version was a few commits newer than v4.10.1.

dennisguse, 2024-01-13:
What you describe sounds possible.
The "position" of a marker is stored in `MarkerColumns.duration` and `.length`.
There is no real link to a TrackPoint as all the data is duplicated.

I am working closely to that problem aka #1725.
Until now there is only an indirect link between Markers and TrackPoints.
The code there is ancient...

dennisguse, 2024-09-08:
Markers are not shown anymore in diagrams: 389c4bed9615fbdf25e242dcd482f7a8b6fbb3bd

closed 2024-09-08 by dennisguse