OpenTracksApp/OpenTracks #1834: Use ORM instead of SQLite SQL

dennisguse, 2024-01-15:
We have quite a lot of code to manage three tables in a SQLite database.
One option would be to use  an ORM like https://developer.android.com/jetpack/androidx/releases/room

Moreover, all DB interaction is done via ContentProvider.
This complicates the situation even further while it has the advantage that all data can be exposed to other applications.

Plan:
- [ ] OpenTracks accesses SQLite DB directly (without CustomContentProvider)
- [ ] Trim down CustomContentProvider to only provide data for Dashboard API (e.g., for OSMDashboard)
- [ ] Check if ORM/Room helps to get rid of the large amount of data handling code

ericjs, 2025-03-14:
I'd suggest also consider using [JOOQ](https://www.jooq.org/) rather than an ORM.

dennisguse, 2025-03-14:
JOOQ is not an Androidx library and OpenTracks only Android standard libraries.

ericjs, 2025-03-14:
Ok, sorry, I saw that you used Java code. I'm not familiar with the limitations of Java on Android.