OpenTracksApp/OpenTracks #1835: Add tested/supported device to cadence/speed sensor lsit

jul-debug, 2024-01-17:
**Is your feature request related to a problem? Please describe.**
Not a feature request but I tested a pair of CYCPLUS C3 (cadence) and S3 (speed) sensors with opentracks and they work. 

Note : sometimes they do not work "out of the box" but will connect if you open the Wahoo fitness app first (non FOSS app, you need to pair them the first time, no actual in app recording required). But I think that Wahoo thing is probably just necessary because of my old bugged phone, and a Foss workaround probably would exist.

dennisguse, 2024-01-18:
You can just edit the README_SENSORS.md :)

jul-debug, 2024-01-18:
Hello Dennis,

First of all thank you for the great software. 

I tried editing the file but apparently I can only fork it. Maybe this is the way to do it but I never used github before so I'm not quite sure what I'm supposed to do... I'm sorry :(

dennisguse, 2024-01-21:
@jul-debug Done :D
Thanks for telling us ;)

closed 2024-01-21 by dennisguse

OMTDesign, 2024-04-20:
I think because the CYCPLUS C3 is capable of recording both speed and cadence, is the reason why they must first be set up first in the wahoo app.

commit bb523b81 (2024-01-21): Document working sensors.

Fixes #1835.