OpenTracksApp/OpenTracks #1836: Option to search settings

ch3thanhs, 2024-01-21:
**Is your feature request related to a problem? Please describe.**
Not exactly a problem but it's related to finding settings quickly.

**Describe the solution you'd like**
The ability to search available options in the settings page.

**Describe alternatives you've considered**
I have no suggestions for alternatives.

**Additional context**
NewPipe uses [this](https://github.com/ByteHamster/SearchPreference) library to implement the search feature, can it be included in OpenTracks?

Thank you for this amazing app!

dennisguse, 2024-01-21:
@ch3thanhs if you have time for this, go ahead. Sounds like a nice feature. 

Anyhow, we need to structure the settings better: we only get more and the current structure does not necessarily fit anymore (eg barometer is in Bluetooth Sensors).

ch3thanhs, 2024-01-22:
@dennisguse, I am not knowledgeable enough to work on it, hopefully, someone picks it up :)

dennisguse, 2024-01-31:
@ch3thanhs any suggestions on how to improve the settings by restructuring?

dennisguse, 2024-05-01:
Too much effort for now...