OpenTracksApp/OpenTracks #1837: [Suggestion] Allow Markers to be used with new tracks

ORIOLESFan02, 2024-01-23:
**Is your feature request related to a problem? Please describe.**
I'd assume so. I thought that markers would automatically be applied to new tracks because I wanted to see how long it takes me to get to that point.

**Describe the solution you'd like**
I would like to see markers that have been previously added to OpenTracks show in new recorded tracks.

**Describe alternatives you've considered**
I don't believe I have any alternatives.

**Additional context**
I walk the same route a lot of the time, and it'd just be cool to see how long it takes for me to get to specific markers that I have previously added. Like I said above, I assumed that the markers would just carry over, assuming I passed them at one point in time during the new track recording.

dennisguse, 2024-01-24:
Where would you like to see the markers?
In some kind of list or on a map (i.e. OSM Dashboard)?

ORIOLESFan02, 2024-01-24:
> Where would you like to see the markers? In some kind of list or on a map (i.e. OSM Dashboard)?

Maybe both OpenTracks and OSM dashboard, again, assuming you pass said marker at least one point during the activity/new track. I hope that makes sense, I'm not the best at explaining things.

pstorch, 2024-01-24:
> I would like to see markers that have been previously added to OpenTracks show in new recorded tracks.

You can already select the previous track with the marker and the currently recorded track in OpenTracks and open OSMDashboard with them. Then you would see the old track with the markers + your new track on one map. Would that already help you?

ORIOLESFan02, 2024-01-24:
> You can already select the previous track with the marker and the currently recorded track in OpenTracks and open OSMDashboard with them. Then you would see the old track with the markers + your new [...]

But I can't get that to work. I'm so confused now.

ORIOLESFan02, 2024-01-24:
Oh I see what you're saying now. That's not something I want to do. It makes it even more confusing. I also would just like to see how long it takes me to get from the starting point to the first marker, and from that first marker to the next one, and so on and so forth.

pstorch, 2024-01-25:
Ok, so you want to import all markers from a previous track into the currently recorded track as something like milestones or checkpoints. In the same order, I guess? 
And the distance to the next shown as air-line distance?

ORIOLESFan02, 2024-01-25:
> Ok, so you want to import all markers from a previous track into the currently recorded track as something like milestones or checkpoints. In the same order, I guess?

Yeah basically. Also it would be cool to see how long it takes to get to said points y'know?

> And the distance to the next shown as air-line distance?

I don't understand that question though.

pstorch, 2024-01-25:
> Also it would be cool to see how long it takes to get to said points y'know?

Based on your average speed? Your current speed? 

> > And the distance to the next shown as air-line distance?
> 
> I don't understand that question though.

OT can't do routing considering roads. It could only show the direct distance through air.

ORIOLESFan02, 2024-01-25:
> Based on your average speed? Your current speed?

Yeah, and time as well. Like the average speed, and then time it takes you to get to the marker/waypoint.

> OT can't do routing considering roads. It could only show the direct distance through air.

I still don't understand :(

dennisguse, 2024-12-19:
I believe this is out of scope - OpenTracks does purely recording and has the explicit feature that each marker belongs to one track. 
If there are use case make it necessary to change this: they need provide user value for more users.

closed 2024-12-19 by dennisguse