OpenTracksApp/OpenTracks #1840: incorrect naming "recording time interval" in the settings

hercek, 2024-01-24:
**Describe the bug**
The setting "GPS" - "Recording time interval" is not correctly named. It should be named "Sampling time interval" or maybe "Reading time interval". Recording into track points happens every "recording distance interval" or about once in 7 seconds when stationary. Well, no clue really. This is how it looks from my experience as a user. I did not check the code.

**Technical information**
 - OpenTracks version: 4.11.0

dennisguse, 2024-01-24:
See GPSManager; it is the update interval passed to Android's LocationManager for the desired interval of updates.

I like 'Sampling time interval' as it is actually not used for recording (in difference to 'Recording distance interval'.
Can you fix this?

hercek, 2024-01-24:
Do you want to have it changed in the translation strings only or also in the source code (i.e. renaming of the relevant java identifiers in the sense \*MinRecordingInterval\* -> \*MinSamplingInterval\*)?

I tried to rename it in the source code as well but not sure it is what you want. The diffstat of this full rename is not too bad:
```
 src/androidTest/java/de/dennisguse/opentracks/services/handlers/GpsStatusTest.java |  4 ++--
 src/main/java/de/dennisguse/opentracks/services/handlers/GPSManager.java           |  6 +++---
 src/main/java/de/dennisguse/opentracks/services/handlers/GpsStatusManager.java     |  4 ++--
 src/main/java/de/dennisguse/opentracks/settings/GpsSettingsFragment.java           | 10 +++++-----
 src/main/java/de/dennisguse/opentracks/settings/PreferencesUtils.java              | 16 ++++++++--------
 src/main/res/values/settings.xml                                                   |  8 ++++----
 src/main/res/values/strings.xml                                                    |  4 +++-
 src/main/res/xml/settings_gps.xml                                                  |  8 ++++----
 8 files changed, 31 insertions(+), 29 deletions(-)
```
The partial rename (only in translations) will change only the ``strings.xml`` and one-line-change in ``GpsSettingsFragment.java``.

dennisguse, 2024-01-25:
Changing the code as well sounds good

PR #1842 (2024-01-25): fix naming for the pace of reading of GPS location in settings
**Link to the the issue**
Fixes #1840

**License agreement**
By opening this pull request, you are providing your contribution under the _Apache License 2.0_ (see [LICENSE.md](LICENSE.md)).
commit 91b5c148 (2024-01-25): fix naming for the pace of reading of GPS location in settings
 Fixes #1840
closed 2024-01-25 by dennisguse