OpenTracksApp/OpenTracks #1848: Opentracks not displaying elevation Loss/Gain

miharashi, 2024-01-31:
Opentracks v4.11.0 no longer displays Loss/Gain on statistics screen. 

To Reproduce
1 Start recording
2 Observe that Loss/Gain remain blank (see screenshot)

Note: lat/lon/elevation are being logged to GPX files

[image]

Device: Galaxy S22 (stock)

dennisguse, 2024-01-31:
Can you check the settings: Bluetooth Sensors -> Barometer and what is set there?
It should be internal, but it likely is not.

miharashi, 2024-01-31:
That was a quick and accurate diagnosis.
I set "Bluetooth Sensors" to "Internal" and Loss/Gain is being displayed.
Thank you. Donation coming.

closed 2024-01-31 by miharashi

LeonPyramid, 2024-08-26:
It seems that I have the same issue, but it seems I cannot select any barometer in the settings.
[image]
I'm on a Redmi Note 9S with Lineageos for microg.

dennisguse, 2024-08-26:
@LeonPyramid the Redmo note has no barometer. 
So, the behavior is correct.

miharashi, 2024-08-26:
I think you sent this to the wrong address. I don't have a Redmo.

(I had a similar issue a while ago and you helped then. Thanks!)