OpenTracksApp/OpenTracks #1850: App crashes after stopping non-GPS workout via Gadgetbridge-linked fitness tracker

beansmachine, 2024-02-06:
**Describe the bug**

After stopping a workout recording via fitness tracker, the app immediately crashes. This issue *doesn't* occur when recording a workout that uses GPS, such as walking or cycling. All requested app permissions are enabled, battery use is unrestricted. 

Crash log:
https://pastebin.com/raw/WNTuKJsP

**To Reproduce**
1. Stop any non-GPS workout on tracker
2. See error

**Technical information**
- Devices: Xiaomi Mi A2 lite, Xiaomi Mi Band 4
- OS: LineageOS 21 (unofficial)
- OpenTracks version: v4.11.0

dennisguse, 2024-02-06:
Can you upload the logs to Github?

dennisguse, 2024-02-06:
Can you give detailed instructed how to reproduce this  (incl. in which if the screen is on/off)?

beansmachine, 2024-02-06:
The crash consistently happens whenever data are sent over from the fitness tracker. Tested this with the app being open, running in background, with the screen on and after unlocking - as soon as the tracker's recording is stopped (including cases when the band says the workout is too short to be recorded), OpenTracks displays the crash log and closes.

dennisguse, 2024-02-06:
Best guess: introduced in 9a0f9271.
When stopping a the service should not be started as it is already running.

closed 2024-02-10 by dennisguse

dennisguse, 2024-02-13:
Fix reverted due to #1853.

And this is actually a duplicate of #1824.
Due to https://github.com/OpenTracksApp/OpenTracks/issues/1824#issuecomment-1941956834

commit ee55be58 (2024-02-10): Bugfix: PublicAPI stop should not start TrackRecordingService in foreground.

Actually, TrackRecordingService will start in foreground if needed.

Fixes #1850.
PR #1851 (2024-02-10): Bugfix: PublicAPI stop should not start TrackRecordingService in fore…
…ground.

Actually, TrackRecordingService will start in foreground if needed.

Fixes #1850.
commit da3dd29b (2024-02-10): Bugfix: PublicAPI stop should not start TrackRecordingService in foreground.

Actually, TrackRecordingService will start in foreground if needed.

Fixes #1850.