OpenTracksApp/OpenTracks #1854: Update README, F-Droid and Play Store screenshots

ch3thanhs, 2024-02-13:
I would like to update the screenshots to reflect the new design of the app, I have created a sample (attached below) and will create the rest if you can provide me with the appropriate screenshots, Let me know which one you prefer.

## Solid Color -

[image]

## With Gradient - 

[image]

dennisguse, 2024-02-13:
That sounds good!
I personally prefer the 2nd version.
Just create a PR and we can merge this :D

There are some templates for these images in `screenshots-svg` and there is the script `UPDATE_SCREENSHOTS.sh`.
This may be a good starting point for some automation.

ch3thanhs, 2024-02-14:
Done :)

closed 2024-02-14 by dennisguse

dennisguse, 2024-03-01:
@ch3thanhs you are right the images were still the old ones.
Fastlane is only working for F-Droid; the PlayStore happens (once a year) manually by me and it has the tendency to not work aka provide weird error messages.

I just did the upload of the images + feature graphic manually and it now awaits PlayStore review.
Thanks for reminding me :+1:

commit af6cf508 (2024-02-14): Update screenshots with new UI (#1857)

Update screenshots with new UI

Fixes #1854.