OpenTracksApp/OpenTracks #1862: Ver 4.11.1 stops recording soon after starting

PeteMacca, 2024-02-16:
As soon as I start recording, it stops. I suspect the GPS is involved. If I start recording and do not move, it keeps recording. If I move just 5 metres, it stops. No error messages. No crash.

...

**Checklist**

- [x ] Consulted [https://dontkillmyapp.com/]
- [x ] Power management: battery optimization is disabled for OpenTracks

**Technical information**

- Device: Pixel 3a
- OS: Android 12
- OpenTracks version: v4.11.1
- OpenTracks commit id (only for nightly builds): 5679

furai, 2024-02-16:
Most likely the same issue as #1863.
I'm affected by this bug as well.

**Technical information**
 - Device: OnePlus 9 Pro
 - OS: Lineageos 20
 - OpenTracks version: v4.11.1

dennisguse, 2024-02-16:
Duplicate of #1853.
Solutions: upgrade to v4.11.2 or downgrade to v4.11.0

BosqueRider, 2024-02-17:
Same problem with v4.11.1, track recording  stops almost immediately.  Downgrade to v4.11.0 was successful.
Device: BLU B130DL
OS: Android 10

Firesphere, 2024-02-17:
When will 4.11.2 appear in FDroid?

dennisguse, 2024-02-17:
@Firesphere I don't know; likely in the next days (process is not really predictable).
But v4.11.2 was already build: https://monitor.f-droid.org/builds/log/de.dennisguse.opentracks/5683#site-footer
... and now awaits signing and publication.

dennisguse, 2024-02-18:
v4.11.2 is now available via F-Droid.

closed 2024-02-18 by dennisguse