OpenTracksApp/OpenTracks #1873: Elevetation for past workouts

DavidHenryThoreau, 2024-02-27:
**Is your feature request related to a problem? Please describe.**
At the moment workouts are display like this picture : 
https://raw.githubusercontent.com/OpenTracksApp/OpenTracks/main/fastlane/metadata/android/en-US/images/phoneScreenshots/screenshot5.png

**Describe the solution you'd like**
I would like to add elevation gain.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

dennisguse, 2024-02-28:
Similar to #937

dennisguse, 2024-04-01:
@DavidHenryThoreau why actually?

DavidHenryThoreau, 2024-04-02:
Because I can't see elevation.