OpenTracksApp/OpenTracks #1874: Import from CSV

DavidHenryThoreau, 2024-02-27:
**Is your feature request related to a problem? Please describe.**
I'm storing workouts stats in personnal app, so I'd like to switch to OpenTracks, so I've made a CSV export and would like to import that CSV to OpenTracks

**Describe the solution you'd like**
How can I import my CSV to OpenTracks

**Describe alternatives you've considered**
Can you add a CSV importer in settings ?

**Additional context**
Add any other context or screenshots about the feature request here.

dennisguse, 2024-02-28:
CSV is a very generic format....
The main problem is that CSV is just a table that has no standard - this is also the reason why OpenTracks exports CSV (in it's definition, but doesn't import it).

Can you not use some scripts to convert those to KML / GPX?
Like https://www.gpsbabel.org/htmldoc-development/fmt_unicsv.html

There are even webpages for this task. 
The other question is if you really want to upload your data onto somebody's webserver.

https://www.gpsvisualizer.com/convert_input

closed 2024-02-28 by dennisguse

DavidHenryThoreau, 2024-02-29:
CSV is format like that : 

```
id,uuid,user_id,sport_id,title,gpx,creation_date,modification_date,workout_date,duration,pauses,moving,distance,min_alt,max_alt,descent,ascent,max_speed,ave_speed,bounds,map,map_id,weather_start,weather_end,notes
39,3958a0ec-f780-4579-b7a4-28deffdd8fe5,3,14,Mont Mc,,2018-08-13,,2018-08-13,7:30:00,,7:30:00,20,,3169,,,,,,,,,,-Salomon
```

DavidHenryThoreau, 2024-02-29:
I'm gonna read about your link if I can use these tools

dennisguse, 2024-03-01:
The CSV format is just the summary - e.g., it contains no coordinates.
So, none of these converters would work...
However, the CSV has a column called **GPX**: could be that the app you used for recordings also exported GPX?
... the data may be in separate files.
And which app is it?

DavidHenryThoreau, 2024-03-04:
I've developed that app and made CSV export in order to import to FitTrackee database structure, but I can change CSV export, if you can post columns order

dennisguse, 2024-03-05:
Which app? Do you have a link?

And OpenTracks only supports importing actual recording (a list of points with timestamp + location or distance traveled); importing summaries alone is not supported.
What you have shown here: https://github.com/OpenTracksApp/OpenTracks/issues/1874#issuecomment-1970680260
... is only a _summary of a recording_ (aka overall statistics).