OpenTracksApp/OpenTracks #1880: Crash for missing permissions does not list all permissions

Mats391, 2024-03-11:
**Describe the bug**
As stated in #1781, #1845 and #1867 you now need to give permission manually for recording to work. The crash message only states permissions for location and nearby devices. However if permission for notifications is not granted, it triggers the same crash with same message. 

**To Reproduce**
1. Enable location + nearby devices permissions
2. Disable notification permission
3. Start OpenTracks and start recording
4. Crash
5. Grant the notification permission
6. Recording works

**Workaround**
If you do not want notification from OpenTracks, you can grant the general permission and the remove the individual notification permissions. Recording will then work, but you wont get the notifications.

**Technical information**
 - Device: Pixel6a
 - OS: CalyxOs 5.4.1
 - OpenTracks version: v4.11.3

Mats391, 2024-03-14:
Do you mean a pullrequest? I did not change any code, the mentioned workaround is how I set up my permissions since I do not want notifications.
[image]

dennisguse, 2024-03-14:
To update the message :)

dennisguse, 2024-04-01:
I cannot reproduce the issue on an Android 13.
I revoked the notification permission and OpenTracks asked again (I rejected). 
No crash though.

commit 48baece0 (2024-04-01): Bugfix: POST_NOTIFICATION permission is optional.

Affected only SDK >= 33.

Fixes #1880.
closed 2024-04-01 by dennisguse